themperek/cocotb-test #4: Recompile if change only

themperek, 2019-06-18:
Based on timestamps: https://stackoverflow.com/questions/22630756/how-does-makefile-know-that-a-file-changed-and-then-recompile-it

StefanD986, 2019-07-25:
Maybe another way to realize that could be: 
Instead of specifying sources, toplevel module, etc in the `run` method create a `HdlCompilationUnit` class that represents this information. Instantiating the class will trigger compilation of the HDL sources (or could be actively triggered by some method call).

Benefit would be that this leaves it up to the user to decide when the files have to be recompiled.

It would also fit in nicely with pytest's [*fixture* scopes](http://doc.pytest.org/en/latest/fixture.html#scope-sharing-a-fixture-instance-across-tests-in-a-class-module-or-session). By creating the `HdlCompilationUnit` within a fixture, and specifying the scope of the fixture one could easily define whether the compiled sources should be reused accross test methods.

(Admittetly `HdlCompilationUnit` is a shitty name, I couln't quickly come up with a better name)

themperek, 2019-07-28:
@StefanD986  I guess need something simple to start with and then can do more advanced. It actually should be very easy to extend this in a way you describe.

themperek, 2019-08-02:
Basic functionality is somehow implemented now (possible). There is a bit of detail and to do this for all simulators. Simulator class is quite independent and can be used similar to what @StefanD986 suggest. 
For all those separate issues should be made.

closed 2019-08-02 by themperek