themperek/cocotb-test #6: GHDL integration

bth105, 2019-06-22:
Hi,
This is exactly what i was looking for. Is there anyway it can be integrated into GHDL. I don't mind helping out im just not that experienced with python.
Kind regards,
Ben

cmarqu, 2019-06-22:
Hi Ben,

supporting ghdl shouldn't be hard, it just needs a new ``_run_ghdl``function in https://github.com/themperek/cocotb-test/blob/master/cocotb_test/run.py similar to the existing ``_run_*``functions, with the options etc. taken from https://github.com/potentialventures/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.ghdl

themperek, 2019-06-28:
Some basic functionally should work now.

There actually seem some bug in cocotb itself since dff example (https://github.com/potentialventures/cocotb/tree/master/examples/dff) does not work with all simulators I have.

closed 2019-06-28 by themperek