themperek/cocotb-test #7: Windows support issues

themperek, 2019-06-23:
This is not so easy considering how setuptools work.
The problem is this: https://stackoverflow.com/questions/34689210/error-exporting-symbol-when-building-python-c-extension-in-windows
The ugly workaround does not seem to work for c++ library (libgpi).

One solution is to call gcc directly. At the end this will be cleanest. Since we will need some dependency function for compiling verilog this can be reused.

themperek, 2019-06-24:
Very ugly workaround but seems to work:
```python
from distutils.command.build_ext import build_ext as _du_build_ext
from unittest.mock import Mock

mockobj = _du_build_ext
mockobj.get_export_symbols = Mock(return_value=None)
```
Will use this.

themperek, 2019-06-24:
Any other ide?

eric-wieser, 2019-06-26:
I think the problem here is that `Extension` is being misused to build shared libraries that are not for direct python consumption at all.

Sharing code between `Extension` modules is actually troublesome in itself. A good example in the python ecosystem is `numpy` and `scipy`. ` numpy` provides a python function that returns a `PyCapsule` containing a table of function pointers, which `scipy` can then use with the numpy header.

This is [the approach recommended in the python docs](https://docs.python.org/3/extending/extending.html#providing-a-c-api-for-an-extension-module)

I don't think this makes a huge amount of sense for use in cocotb though, since there are no intended consumers of the intermediate libraries.

themperek, 2019-06-26:
We may need to finish with gcc wrapper (this is not too scary) will see.

closed 2019-06-26 by themperek