themperek/cocotb-test #13: USER API

themperek, 2019-06-27:
Requirements:
- simple and compatible with all tools for basic usage
- possible to derive custom configuration/custom simulation for advanced usage

Possible idea:  A generic simulation class with attributes for settings/parameters and build()/run() functions that can be derived and customized. Can be used as an input parameter to test function. 

Use the exixting API?
 - https://github.com/olofk/edalize
 - https://vunit.github.io/
..

themperek, 2019-06-28:
Which one?

```python
def custom_sim( ... ):
    #do something

test_defaut():
    run( ... ) 

test_custom():
    run( ..., build=custom_sim)
```

or

```python
from cocotb_test.sim import icarus

class custom_sim(icarus):
    def build(self):
        #do something
        self.verilog_sources 

test_defaut():
    run( sim(...) ) 

test_custom():
    run ( custom_sim( ... ) ) 
```

or prebuild libs (in setup.py or command tool or some pytest fixture)

```python

from cocotb_test.sim import sim

def custom_sim( ... ):
    #do something

test_defaut():
    sim( ... ) 

test_custom():
    custom_sim( ... ) 
```

any other?

themperek, 2019-06-28:
Something to prebuild libraries.
```python
@pytest.fixture(scope="module", autouse=True)
def my_fixture():
    print ('INITIALIZATION')
    yield param
    print ('TEAR DOWN')
```

themperek, 2019-06-30:
```python
import subprocess
import os

class Simulator:
    def __init__(
        self,
        run_dir,
        sim_dir,
        lib_dir,
        lib_ext,
        toplevel,
        verilog_sources=[],
        vhdl_sources=[],
        includes=[],
        defines=[],
        extra_compile_args=[],
        extra_simulation_args=[],
        **kwargs
    ):

        self.run_dir = run_dir
        self.sim_dir = sim_dir
        self.lib_dir = lib_dir
        self.lib_ext = lib_ext

        self.toplevel = toplevel
        self.verilog_sources = get_abs_paths(verilog_sources)
        self.vhdl_sources = get_abs_paths(vhdl_sources)
        self.includes = get_abs_paths(includes)

        self.defines = defines
        self.extra_compile_args = extra_compile_args
        self.extra_simulation_args = extra_simulation_args

        for arg in kwargs:
            setattr(self, arg, kwargs[arg])

    def build_command(self):
        pass

    def run(self):
        cmds = self.build_command()
        self.execute(cmds)

    def get_include_commands(self, includes):
        pass

    def get_define_commands(self, defines):
        pass

    def get_abs_paths(self, paths):
        paths_abs = []
        for path in paths:
            if os.path.isabs(path):
                paths_abs.append(path)
            else:
                paths_abs.append(os.path.abspath(os.path.join(self.run_dir, path)))

        return get_abs_paths

    def execute(self, cmds):
        for cmd in cmds:
            print(" ".join(cmd))
            process = subprocess.check_call(cmd)

class Icarus(Simulator):
    def get_include_commands(self, includes):
        include_cmd = []
        for dir in includes:
            include_cmd.append("-I")
            include_cmd.append(dir)

        return include_cmd

    def get_define_commands(self, defines):
        defines_cmd = []
        for define in defines:
            defines_cmd.append("-D")
            defines_cmd.append(define)

        return defines_cmd

    def build_command(self):

        sim_compile_file = os.path.join(self.sim_dir, "sim.vvp")

        cmd_compile = (
            [
                "iverilog",
                "-o",
                sim_compile_file,
                "-D",
                "COCOTB_SIM=1",
                "-s",
                self.toplevel,
                "-g2012",
            ]
            + self.get_define_commands(self.defines)
            + self.get_include_commands(self.includes)
            + self.verilog_sources
        )

        cmd_run = ["vvp", "-M", self.lib_dir, "-m", "gpivpi", sim_compile_file]

        return [cmd_compile, cmd_run]

```

cmarqu, 2019-07-01:
My use case would be to say e.g. ``cmd_run = "xrun -loadvpi libvpi -f xrun.f"`` and everything that is not cocotb-specific (like the source files, toplevel, simdir) comes from the ``xrun.f`` where it is defined in a format that the simulator understands.

themperek, 2019-07-01:
In this case ift would look somrhitng like this:

```python
class XrunCmd(Simulator):
    def build_command(self):
        cmd = ["xrun", "-loadvpi", libvpi, "-f", self.cmd_file]
        return [cmd]
```

in test 

```python
tes_xyz():
    run(cmd_file=cmd.f', sim=XrunCmd)
```

How does this look like?

cmarqu, 2019-07-01:
Looks good! I may still need to give some extra information like ``TOPLEVEL_LANG`` (and I guess ``TOPLEVEL`` too) to cocotb, but the most important part is the reuse of all the information in the *.f file (or *.do in the case of Mentor).

closed 2019-07-03 by themperek