themperek/cocotb-test #15: ghdl error

bth105, 2019-07-01:
Hi,
Thanks for putting in ghdl support. I have probably missed something, but when i run the dff test i am getting a error

E           CalledProcessError: Command '['ghdl', '-r', 'dff', '--vpi=/home/ben/tools/cocotb-test/tests/build/libvpi.so']' returned non-zero exit status 1

when i run the command separately i get the following issue
loading VPI module '/home/ben/tools/cocotb-test/build/libvpi.so'
libcocotbutils.so: cannot open shared object file: No such file or directory
ghdl:error: cannot load VPI module

obviously you wrote the test with a verilog dut but i changed it to vhdl

Any advice on how to get it working as i assume i am just missing something

Thanks

Ben

themperek, 2019-07-02:
` rm -rf build` ?

bth105, 2019-07-02:
it looks like it not able to find libs_dir, do i need to setup any environmental variables
cheers

themperek, 2019-07-02:
It worked for me.
Can you share the log file? 

BTW: It all should change quite a bit next days. Few better ideas.

bth105, 2019-07-02:
~/tools/cocotb-test/tests$ SIM=ghdl pytest -s test_dff.py 
=========================================================================================== test session starts ============================================================================================
platform linux2 -- Python 2.7.15rc1, pytest-4.6.4, py-1.8.0, pluggy-0.12.0
rootdir: /home/ben/tools/cocotb-test
collected 1 item                                                                                                                                                                                           

test_dff.py ghdl -a /home/ben/tools/cocotb-test/tests/dff.vhd
ghdl -e dff
ghdl -r dff --vpi=/home/ben/tools/cocotb-test/tests/build/libvpi.so
loading VPI module '/home/ben/tools/cocotb-test/tests/build/libvpi.so'
libcocotbutils.so: cannot open shared object file: No such file or directory
ghdl:error: cannot load VPI module
F

================================================================================================= FAILURES =================================================================================================
_________________________________________________________________________________________________ test_dff _________________________________________________________________________________________________

    def test_dff():
        Run(
            vhdl_sources=['dff.vhd'], # sources
            toplevel='dff',            # top level HDL
            python_search=['.'],       # search path for cocotb test module
>           module='dff_cocotb'        # name of cocotb test module
        )

test_dff.py:8: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../cocotb_test/run.py:556: in Run
    ext_name,
../cocotb_test/run.py:376: in _run_ghdl
    process = subprocess.check_call(cmd, cwd=sim_build_dir)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

popenargs = (['ghdl', '-r', 'dff', '--vpi=/home/ben/tools/cocotb-test/tests/build/libvpi.so'],), kwargs = {'cwd': '/home/ben/tools/cocotb-test/tests/sim_build'}, retcode = 1
cmd = ['ghdl', '-r', 'dff', '--vpi=/home/ben/tools/cocotb-test/tests/build/libvpi.so']

    def check_call(*popenargs, **kwargs):
        """Run command with arguments.  Wait for command to complete.  If
        the exit code was zero then return, otherwise raise
        CalledProcessError.  The CalledProcessError object will have the
        return code in the returncode attribute.
    
        The arguments are the same as for the Popen constructor.  Example:
    
        check_call(["ls", "-l"])
        """
        retcode = call(*popenargs, **kwargs)
        if retcode:
            cmd = kwargs.get("args")
            if cmd is None:
                cmd = popenargs[0]
>           raise CalledProcessError(retcode, cmd)
E           CalledProcessError: Command '['ghdl', '-r', 'dff', '--vpi=/home/ben/tools/cocotb-test/tests/build/libvpi.so']' returned non-zero exit status 1

/usr/lib/python2.7/subprocess.py:190: CalledProcessError
========================================================================================= 1 failed in 0.22 seconds =========================================================================================

Hopefully you dont mind me pasting in the output. Oh sounds cool.

themperek, 2019-07-02:
doing `rm -rf build` does not help?
It works for me on centos7 on python 2.7 and 3.7. I will try to add this to CI.

themperek, 2019-07-02:
+ there is this: https://github.com/potentialventures/cocotb/issues/979
I do not know why.

bth105, 2019-07-04:
so it looks like it was due to LD_LIBRARY_PATH not being set to the location of libcocotbutils.so. 

Yer i have now bumped into the #979 but it seems to be pretty much working now.

Cheers for your help

themperek, 2019-07-04:
This is strange.
There should be no need to set `LD_LIBRARY_PATH`  at this point.
Maybe you have something set locally.

bth105, 2019-07-04:
Oh okay, maybe im masking my actual issue.
how does it pick up the location of the libcocotbutils.so?

themperek, 2019-07-04:
This should do the trick: https://github.com/themperek/cocotb-test/blob/f0d6e48116f6fd2a3d3823638b7e676548958a58/cocotb_test/build_libs.py#L182

Maybe add the same here (to libcocotbutils ) : https://github.com/themperek/cocotb-test/blob/f0d6e48116f6fd2a3d3823638b7e676548958a58/cocotb_test/build_libs.py#L108

Maybe this is OS dependent?

/Tomek

bth105, 2019-07-05:
I'm using ubuntu 18.04. I have fetched (ad pip installed) the latest but the issue still persists. Sorry about this

themperek, 2019-07-06:
Can you try `try_ghdl` branch https://github.com/themperek/cocotb-test/tree/try_ghdl

bth105, 2019-07-07:
tried that branch and it worked. Thanks for sorting this.

commit d4ba22fe (2019-07-06): try to fix #15
closed 2019-07-07 by themperek