themperek/cocotb-test #28: Add support for verilog plusargs for icarus.

StefanD986, 2019-07-19:
The verilog standard defines *plusargs* (command line arguments that start with a `+`), that can be passed to the simulator. These *plusargs* can be retrieved in the verilog code with the `$test$plusargs` and `$value$plusargs` system tasks.

Icarus verilog requires these *plusargs* to be passed **after** the design file name (in the icarus man page the *plusargs* are called `extended-args`):
``` bash
vvp [-inNsvV] [-Mpath] [-mmodule] [-llogfile] inputfile [extended-args...]
```

It might make sense to allow another argument to the `run(...)`  method that takes these additional *plusargs*, and then appends them after `[self.sim_file]` here (without removing the possibility of adding `extra_simulation_args`):

https://github.com/themperek/cocotb-test/blob/32b8014796cd914ff9e6fb16491487b7f5fad49c/cocotb_test/simulator.py#L122-L127

I am not sure how other simulators handle these *plusargs*. Would be good to keep compatibility.

themperek, 2019-07-19:
The idea is to keep the basic code simple and allow easy full customization:

see this: https://github.com/themperek/cocotb-test/blob/master/tests/test_dff_custom_sim.py#L7

I do not know how many features we want in the basic version?

@cmarqu Any opinion?

StefanD986, 2019-07-19:
Maybe a relevant related issue is https://github.com/cocotb/cocotb/issues/711

themperek, 2019-07-19:
I guess I can add this.

cmarqu, 2019-07-19:
I have no strong feelings either way, but it could be said that if you use plusargs you are advanced enough that you can be expected to use the custom setup. We could expand the Icarus example to show the use of plusargs.

themperek, 2019-07-20:
I have added it in #29 for Icarus and Questa.

closed 2019-07-20 by themperek