themperek/cocotb-test #33: Tests randomly fail after 3000+ tests.

StefanD986, 2019-07-24:
I am currently (ab?)using cocotb-test to run a simulation repeatedly, with different test vectors for each run.

With this I am running into an issue, that at some point the subprocess call here fails:
https://github.com/themperek/cocotb-test/blob/e9ae21843c1ec37ca06414479ab683d4c2588471/cocotb_test/simulator.py#L76

With `OSError: [Errno 12] Cannot allocate memory` or `OSError: [Errno 7] Argument list too long` (The latter is a bit misleading, I think it is indeed caused by some failure to allocate memory for the subprocess).

This happens repeatably on my system roughly after about 3000 tests, but I would assume this is highly system(-memory) dependent.

I tried to work around this by calling `gc.collect()` after each call to `run(...)`, in hope that this would free, however without success.

Maybe one way to fix this could be: Instead of launching a process every time, a shell could be launched, which is kept open and wrapped in some sort of *runner* object. This runner object is then used to launch `vvp`, but the process itself is reused.

I must admit this is just a quick thought, and might cause trouble later when using parallized tests (the runner cannot be used by multiple test runs at the same time).

Maybe relevant SO questions:
https://stackoverflow.com/questions/1367373/python-subprocess-popen-oserror-errno-12-cannot-allocate-memory

themperek, 2019-07-24:
Can you try now?

StefanD986, 2019-07-25:
Looks good: I don't see the error with the new changes.

Thanks a lot for the quick fix!

closed 2019-07-25 by StefanD986

marianomds, 2022-08-08:
I am having a very similar issue to the first comment. I am getting a "OSError: [Errno 7] Argument list too long" error when running a regression with about 5000 tests. If I reduce the tests number to about 3000, the error doesn't seem to show up. Any thoughts on how I could avoid this issue? Thanks!

themperek, 2022-08-08:
@marianomds 
I have a feeling you may be having this https://unix.stackexchange.com/questions/336934/raise-128kib-limit-on-environment-variables-in-linux but hard to say without details.

marianomds, 2022-08-10:
Thanks a lot for your answer. Unfortunately, I don't have much more details, but the error messages I am getting are the following:

```
            seed=seed,
            parameters=parameters,
            extra_env=parameters,
            sim_build="sim_build/"
>           + "_".join(("{}={}".format(*i) for i in parameters.items()))
        )

regression.py:22: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
../../../.local/lib/python3.6/site-packages/cocotb_test/simulator.py:941: in run
    return sim.run()
../../../.local/lib/python3.6/site-packages/cocotb_test/simulator.py:217: in run
    self.execute(cmds)
../../../.local/lib/python3.6/site-packages/cocotb_test/simulator.py:263: in execute
    env=self.env
/usr/lib64/python3.6/subprocess.py:729: in __init__
    restore_signals, start_new_session)
```

```
            if issubclass(child_exception_type, OSError) and hex_errno:
                errno_num = int(hex_errno, 16)
                child_exec_never_called = (err_msg == "noexec")
                if child_exec_never_called:
                    err_msg = ""
                    # The error must be from chdir(cwd).
                    err_filename = cwd
                else:
                    err_filename = orig_executable
                if errno_num != 0:
                    err_msg = os.strerror(errno_num)
                    if errno_num == errno.ENOENT:
                        err_msg += ': ' + repr(err_filename)
>               raise child_exception_type(errno_num, err_msg, err_filename)
E               OSError: [Errno 7] Argument list too long: 'xrun'

/usr/lib64/python3.6/subprocess.py:1364: OSError
```

marianomds, 2022-08-11:
I found the issue, and it was not related to cocotb-test (I was mistakenly adding a folder to the python path at each test run, collapsing the memory after ~3000 tests). Thanks a lot for your help!

PR #34 (2019-07-24): Check if lib_dir path PATH before adding
Should fix #33 33