themperek/cocotb-test #40: tkinter dependency

qarlosalberto, 2019-08-08:
Hello,

I have had a problem with tkinter dependency in Linux Mint 19 Cinnamon:

```
E   ModuleNotFoundError: No module named 'tkinter'
```
I have solved it with:
```
sudo apt install -y python3-tk
```
Maybe you could specify it in the README.md.

Thank you, nice software ;)

themperek, 2019-08-09:
@eric-wieser Any idea how to do this other way? Maybe copy part of this code?

eric-wieser, 2019-08-09:
This "dependency" is a built-in python module that Ubuntu has decided to separate into a separate package. Since the package is managed at the is level and not the pip level, I don't know how to solve this in the setup.py

Vendoring the function would be totally reasonable. I think the license is fine.

StefanD986, 2019-08-12:
I also ran into this issue in the past, but with compiled-from-source python. If I remember correctly, the tkinter module is not built with default compile options.

It was not a big deal for me to re-compile, however since its just about one function from the tkinter module I'd also vote for copy-pasting this function if licensing permits it.

themperek, 2019-08-12:
I guess for us this is only important:

https://github.com/MacPardo/PsiturkFlanker/blob/1f1f19e673283c3d055d6ff6f2fb6147a6f1a301/psiturk/lib/python2.7/lib-tk/Tkinter.py#L85-L91

themperek, 2019-08-17:
@eric-wieser Can you give me an example where this is needed?

eric-wieser, 2019-08-22:
> I guess for us this is only important

Or to use a more up-to-date source, https://github.com/python/cpython/blob/71876fa438f706b211360d8c205cb985906212ee/Lib/tkinter/__init__.py#L52-L83

> Can you give me an example when this is needed?

Of when stringify is needed? The simple answer is that there is exactly one correct way to send strings between languages, and that is to use appropriate escape sequences - just like you would `html.escape` to output a python string to an html paragraph, or `json.dumps` to output a python string to json.

Given this premise, there are three choices:

1. Use the (private-ish) function present from the standard library that does this escaping. Unfortunately Ubuntu decided to create their own definition of standard library
2. Copy and paste such a function from a reputable source (such as the cpython source code), and include it internally
3. Write our own escaping function by following the TCL spec.

`master` contains 1. I have no qualms with moving to 2, since `python-tk` is a somewhat heavy and unnecessary dependency.

closed 2019-09-28 by themperek