themperek/cocotb-test #41: Error ghdl + python 2.7

qarlosalberto, 2019-08-12:
I am running a simple example and I have this problem:

```
==================================================================================== test session starts =====================================================================================
platform linux2 -- Python 2.7.15+, pytest-3.3.2, py-1.5.2, pluggy-0.6.0
rootdir: /home/cruiz/Descargas/cocotbExamples/simple, inifile:
collected 1 item                                                                                                                                                                             

pytest_adder.py ghdl -i /home/cruiz/Descargas/cocotbExamples/simple/adder.vhd
ghdl -m adder
analyze /home/cruiz/Descargas/cocotbExamples/simple/adder.vhd
elaborate adder
ghdl -r adder --vpi=/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so
loading VPI module '/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so'
/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so: cannot open shared object file: No such file or directory
./adder:error: cannot load VPI module
F                                                                                                                                                                      [100%]

========================================================================================== FAILURES ==========================================================================================
______________________________________________________________________________________ test_adder_vhdl _______________________________________________________________________________________

    @pytest.mark.skipif(os.getenv("SIM") == "icarus", reason="VHDL not suported")
    def test_adder_vhdl():
>       run(vhdl_sources=["adder.vhd"], toplevel="adder", module="test_adder", toplevel_lang="vhdl")

pytest_adder.py:7: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/usr/local/lib/python2.7/dist-packages/cocotb_test/run.py:35: in run
    results_xml_file = sim.run()
/usr/local/lib/python2.7/dist-packages/cocotb_test/simulator.py:105: in run
    self.execute(cmds)
/usr/local/lib/python2.7/dist-packages/cocotb_test/simulator.py:133: in execute
    process = subprocess.check_call(cmd, cwd=self.sim_dir, env=self.env)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

popenargs = (['ghdl', '-r', 'adder', '--vpi=/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so'],)
kwargs = {'cwd': '/home/cruiz/Descargas/cocotbExamples/simple/sim_build', 'env': {'CINNAMON_VERSION': '4.0.10', 'COCOTB_RESULTS_FILE_NAME': '/tmp/tmpg1hkuX', 'COCOTB_SIM': '1', 'COLORTERM': 'truecolor', ...}}
retcode = 1, cmd = ['ghdl', '-r', 'adder', '--vpi=/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so']

    def check_call(*popenargs, **kwargs):
        """Run command with arguments.  Wait for command to complete.  If
        the exit code was zero then return, otherwise raise
        CalledProcessError.  The CalledProcessError object will have the
        return code in the returncode attribute.
    
        The arguments are the same as for the Popen constructor.  Example:
    
        check_call(["ls", "-l"])
        """
        retcode = call(*popenargs, **kwargs)
        if retcode:
            cmd = kwargs.get("args")
            if cmd is None:
                cmd = popenargs[0]
>           raise CalledProcessError(retcode, cmd)
E           CalledProcessError: Command '['ghdl', '-r', 'adder', '--vpi=/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/libvpi.so']' returned non-zero exit status 1

/usr/lib/python2.7/subprocess.py:190: CalledProcessError
```
My OS is: Linux Mint 19.1
cocotb-test: 0.0.4.dev1

qarlosalberto, 2019-08-12:
I have the same error with python3.

themperek, 2019-08-12:
Thank you.

Seems to work with conda: https://dev.azure.com/themperek/themperek/_build/results?buildId=326

Can you do: `pip uninstall cocotb-test` and then "pip install` with `-v`. Is there anything in `/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/ghdl/` ?

qarlosalberto, 2019-08-12:
It's empty:

```
cruiz@DA:~/Descargas/out$ ls /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/
icarus  questa
```
I have an error in the installation:
```
    building 'libvpi' extension
    creating /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DVPI_CHECKING=1 -I/usr/local/lib/python2.7/dist-packages/cocotb/share/include -I/usr/include/python2.7 -c /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiImpl.cpp -o /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiImpl.o
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DVPI_CHECKING=1 -I/usr/local/lib/python2.7/dist-packages/cocotb/share/include -I/usr/include/python2.7 -c /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiCbHdl.cpp -o /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiCbHdl.o
    x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiImpl.o /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/vpi/VpiCbHdl.o -L/usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa -lgpi -lgpilog -o /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/libvpi.so -Wl,-rpath,$ORIGIN
    building 'libfli' extension
    creating /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/fli
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/lib/python2.7/dist-packages/cocotb/share/include -I/usr/include -I/usr/include/python2.7 -c /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/fli/FliImpl.cpp -o /usr/local/lib/python2.7/dist-packages/cocotb_test/libs/questa/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/fli/FliImpl.o
    In file included from /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/fli/FliImpl.cpp:32:0:
    /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/fli/FliImpl.h:32:10: fatal error: mti.h: No existe el archivo o el directorio
     #include "mti.h"
              ^~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-28s6XZ/cocotb-test/setup.py'"'"'; __file__='"'"'/tmp/pip-install-28s6XZ/cocotb-test/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-obQ_yB/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
```

themperek, 2019-08-12:
On ubunu 18.04 (docker) : `Linux 835128a2d1d6 3.10.0-957.21.2.el7.x86_64 #1 SMP Wed Jun 5 14:26:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

Python: `Python 2.7.15+`

```GHDL 0.36 (v0.36) [Dunoon edition]
 Compiled with GNAT Version: 7.4.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

themperek, 2019-08-12:
Ok.
It will try to compile FLI interface if you have `vsim` executable. Do you have Modelsim/Questa installed? 
Setup does not fail in general?  
Maybe should catch this expectation and continue?

qarlosalberto, 2019-08-12:
Yes, I have ModelSim Starter Edition installed.
The installation is finished and cocotb-test is installed, but the ghdl folder is empty.
I have the same error with: test_dff_vhdl
It runs with Icarus.

I don't know what the error is. On my personal computer it worked ... :/

qarlosalberto, 2019-08-12:
I attach the output:
```
ERROR: Command errored out with exit status 1:
   command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rqjIEu/cocotb-test/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rqjIEu/cocotb-test/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-Z9LKO8 --python-tag cp27
       cwd: /tmp/pip-install-rqjIEu/cocotb-test/
  Complete output (116 lines):
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/cocotb_test
  copying cocotb_test/build_libs.py -> build/lib.linux-x86_64-2.7/cocotb_test
  copying cocotb_test/simulator.py -> build/lib.linux-x86_64-2.7/cocotb_test
  copying cocotb_test/__init__.py -> build/lib.linux-x86_64-2.7/cocotb_test
  copying cocotb_test/cli.py -> build/lib.linux-x86_64-2.7/cocotb_test
  copying cocotb_test/run.py -> build/lib.linux-x86_64-2.7/cocotb_test
  installing to build/bdist.linux-x86_64/wheel
  running install
  running install_lib
  creating build/bdist.linux-x86_64
  creating build/bdist.linux-x86_64/wheel
  creating build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data
  creating build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib
  creating build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  copying build/lib.linux-x86_64-2.7/cocotb_test/build_libs.py -> build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  copying build/lib.linux-x86_64-2.7/cocotb_test/simulator.py -> build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  copying build/lib.linux-x86_64-2.7/cocotb_test/__init__.py -> build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  copying build/lib.linux-x86_64-2.7/cocotb_test/cli.py -> build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  copying build/lib.linux-x86_64-2.7/cocotb_test/run.py -> build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test
  running install_egg_info
  running egg_info
  writing requirements to cocotb_test.egg-info/requires.txt
  writing cocotb_test.egg-info/PKG-INFO
  writing top-level names to cocotb_test.egg-info/top_level.txt
  writing dependency_links to cocotb_test.egg-info/dependency_links.txt
  writing entry points to cocotb_test.egg-info/entry_points.txt
  reading manifest file 'cocotb_test.egg-info/SOURCES.txt'
  writing manifest file 'cocotb_test.egg-info/SOURCES.txt'
  Copying cocotb_test.egg-info to build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test-0.0.3.egg-info
  running install_scripts
  building 'libcocotbutils' extension
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages/cocotb
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages/cocotb/share
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages/cocotb/share/lib
  creating /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/utils
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/lib/python2.7/dist-packages/cocotb/share/include -I/usr/include/python2.7 -c /usr/local/lib/python2.7/dist-packages/cocotb/share/lib/utils/cocotb_utils.c -o /tmp/pip-install-rqjIEu/cocotb-test/build/bdist.linux-x86_64/wheel/cocotb_test-0.0.3.data/purelib/cocotb_test/libs/icarus/usr/local/lib/python2.7/dist-packages/cocotb/share/lib/utils/cocotb_utils.o
  x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-MW0004/python2.7-2.7.15=. -fstack-protector-strong -Wf [... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rqjIEu/cocotb-test/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rqjIEu/cocotb-test/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-dv0QLy/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
```

themperek, 2019-08-12:
You have `vsim` executable and it tries to compile FLI interface (but for some reason cannot find `mti.h` that is part of Modelsim. I can try to work this around but it is interesting why ModelSim FLI fails.

It is here: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/build_libs.py#L271-L289

Where is your `vsim`?

qarlosalberto, 2019-08-12:
Yes, it is a problem with ModelSim. If I rename vsim to xvsim, the installation works.

vsim is in: /opt/modelsim/18.0/modelsim_ase/linuxaloem/

My ModelSim version: 
```
Model Technology ModelSim ALTERA STARTER EDITION vsim 10.6c Simulator 2017.07 Jul 26 2017
```

qarlosalberto, 2019-08-12:
ModelSim ALTERA STARTER EDITION has not FLI, can it be the problem?

themperek, 2019-08-12:
>ModelSim ALTERA STARTER EDITION has not FLI, can it be the problem?

Probably. Will guard this part so it does not fail everything.

themperek, 2019-08-13:
It should be fixed now.  It should not fail setup if cannot compile FLI.
If did not fix let me know/reopen.

closed 2019-08-13 by themperek