themperek/cocotb-test #42: [Feature] Output object return

qarlosalberto, 2019-08-12:
It might be interesting if run() returns an object similar to results.xml.

themperek, 2019-08-12:
It does already.

closed 2019-08-12 by themperek

qarlosalberto, 2019-08-12:
My fault. Incomprehensibly I have not seen it.

themperek, 2019-08-12:
Be aware if you run many tests it will be overwritten every time. 
You can use  `pip install https://github.com/themperek/cocotb/archive/results_xml.zip` to get new file every test.
https://github.com/cocotb/cocotb/pull/1053

qarlosalberto, 2019-08-12:
Mmm.. And is there any way to show the tests to pass different settings? In VUnit is:

```
g_SIZE_INPUT  = [10,20,30]

tb_generated = adder_top_tb_lib.entity("adder_top_tb")

for test in tb_generated.get_tests():
    for i in range(0,len(g_SIZE_INPUT)):
        test.add_config(name="test_input_"+str(g_SIZE_INPUT[i]),
                        generics=dict(g_SIZE_INPUT=g_SIZE_INPUT)
        )
```
An example https://github.com/qarlosalberto/cordicHDL/blob/develop/tb/cordic_top_run.py [line:144-156]

themperek, 2019-08-12:
Did not implment generics (can add) or can use : `compile_arg, simulation_args,extra_args)` or `plus_args`.

There is a standard way in pytest/python to for this (test parametryzation):

```python
@pytest.mark.parametrize("plus_arg", range(100))
def test_plus_args(plus_arg):
    run(..., plus_arg)
```

The advantage is this can run in parallel (in principle) and reporting.

qarlosalberto, 2019-08-12:
Nice!!

qarlosalberto, 2019-08-13:
I would be nice add the generics. Im doing it manually:

https://github.com/qarlosalberto/cocotbExamples/blob/master/tools/cocotb/simple/cocotest_test_adder.py

themperek, 2019-08-14:
Something like this is more python like.
```python
DATA_WIDTH = 5
@pytest.mark.parametrize("DATA_WIDTH", range(2,DATA_WIDTH))
def test_adder_vhdl(DATA_WIDTH):
	run(vhdl_sources=["../../../hdl/adder.vhd"],
		simulation_args=["--vcd=func.vcd","-gDATA_WIDTH="+str(DATA_WIDTH)],
		toplevel="adder",
		module="test_adder",
		toplevel_lang="vhdl"
	)
```

qarlosalberto, 2019-08-14:
And, would it be possible to open the API to accept generics? Similar to:

```python
DATA_WIDTH = 5
@pytest.mark.parametrize("DATA_WIDTH", range(2,DATA_WIDTH))
def test_adder_vhdl(DATA_WIDTH):
	run(vhdl_sources=["../../../hdl/adder.vhd"],
		simulation_args=["--vcd=func.vcd",
		toplevel="adder",
                generics=[DATA_WIDTH],
		module="test_adder",
		toplevel_lang="vhdl"
	)
```