themperek/cocotb-test #44: Checkpointing

mciepluc, 2019-08-21:
Does it make sense to include [checkpointing](https://github.com/mciepluc/apbi2c_cocotb_example/blob/master/tb/checkpoint.py) functionality  as the part of this package?
If cocotb-test is going to be merged with core cocotb, it is probably too much complexity to be added (also including parallel regression specific functions).

themperek, 2019-08-21:
I would think it can be part of cocotb-coverage for now and go directly to cocotb (but should be better integrated somehow).

themperek, 2019-10-19:
I think it has to go into cocotb so this is load/saved before/after running a test. I have no control here.

closed 2019-10-19 by themperek

themperek, 2019-10-19:
Maybe we do a page for this?

Would just have `load` and `save` function.