themperek/cocotb-test #46: SIM env needed when simulator is hardcoded

andresdemski, 2019-08-23:
Hi @themperek 

When i'm trying to hardcode the simulator (no SIM env) i get the following error:
```python
run(simulator=Icarus,
    toplevel=name,
    module=module,
    verilog_sources=[verilog_file])
```

```
Traceback (most recent call last):
  File "test.py", line 15, in <module>
    test_module()
  File "test.py", line 12, in test_module
    run(design, get_current_module(), ports=ports, vcd_file='output.vcd')
  File "/home/ademski/Documents/opensource/nmigen-cocotb/nmigen_cocotb.py", line 73, in run
    compile_args=compile_args_waveforms if vcd_file else [])
  File "/home/ademski/Documents/opensource/nmigen-axi/venv/lib/python3.7/site-packages/cocotb_test/run.py", line 22, in run
    sim = simulator(**kwargs)
  File "/home/ademski/Documents/opensource/nmigen-axi/venv/lib/python3.7/site-packages/cocotb_test/simulator.py", line 179, in __init__
    super(Icarus, self).__init__(*argv, **kwargs)
  File "/home/ademski/Documents/opensource/nmigen-axi/venv/lib/python3.7/site-packages/cocotb_test/simulator.py", line 37, in __init__
    self.lib_dir = os.path.join(libs_dir, os.getenv("SIM"))
  File "/usr/lib/python3.7/posixpath.py", line 94, in join
    genericpath._check_arg_types('join', a, *p)
  File "/usr/lib/python3.7/genericpath.py", line 149, in _check_arg_types
    (funcname, s.__class__.__name__)) from None
TypeError: join() argument must be str or bytes, not 'NoneType'
```

I workaround it with a simple `os.environ['SIM'] = 'icarus'` before the `run(...)`. I would be great that if `simulator` argument is present, it won't need the SIM env.

themperek, 2019-08-23:
You can do this: https://github.com/themperek/cocotb-test/blob/master/tests/test_dff_custom_sim.py#L24-L31

But in general, I think this can be done better.
Ideas welcome here: https://github.com/themperek/cocotb-test/issues/43

themperek, 2019-08-26:
```python
from cocotb_test.simulator import Icarus

run(simulator=Icarus,
    toplevel=name,
    module=module,
    verilog_sources=[verilog_file])
```

closed 2019-09-19 by themperek