themperek/cocotb-test #49: Simulation pseudo-fails

StefanD986, 2019-09-16:
I am using parametric tests to re-run a simulation with different inputs, in parallel using pytest xdist.

In some cases, the simulation will report as failed, this is the stacktrace:

```python
>       assert os.path.isfile(results_xml_file), "Simulation terminated abnormally. Results file not found."
E       AssertionError: Simulation terminated abnormally. Results file not found.
E       assert False
E        +  where False = <function isfile at 0x7fecdc42ea60>('/tmp/tmpd5zd6huo')
E        +    where <function isfile at 0x7fecdc42ea60> = <module 'posixpath' from '/usr/lib/python3.6/posixpath.py'>.isfile
E        +      where <module 'posixpath' from '/usr/lib/python3.6/posixpath.py'> = os.path

../../cocotb-test/cocotb_test/run.py:36: AssertionError
```

I don't really have a reliable way to reproduce it, However I assume its some race condition caused by the parallelized tests, because I don't see that in case I don't use parallelization.

themperek, 2019-09-16:
Probably the patch for this is in cocotb (master) already I will push change to cocotb-test asap.

themperek, 2019-09-16:
Can you try now?

StefanD986, 2019-09-16:
Currently not in the office, but will try first thing in the morning. Thanks a lot!

themperek, 2019-09-16:
You need to use current master for cocotb

StefanD986, 2019-09-17:
Yes, its fixed with current master of cocotb and cocotb-test. Thanks again!

closed 2019-09-17 by StefanD986