themperek/cocotb-test #57: Way to pass parameters to the cocotb test

benreynwar, 2019-10-22:
Is there a way to pass parameters to the cocotb test?  Typically I'll have a large number of parameters that I use to generate my vhdl.  The cocotb test needs access to these parameters to determine to properly test the core.

One option would be to add a generic parameter to the top level module that is just a string pointing at a file that contains all the parameters that the test will need (i.e. dut.TEST_PARAMETERS_FILENAME), but that seems pretty ugly.

themperek, 2019-10-23:
In current cocotb test you can pass a environmental variable ex.:
```python
run(.., extra_env={PARAM1:"a", PARAM2:"b"})
```

In the test you can get it with:
```python
os.getenv("PARAM1", "default")
```

Not sure if this answer your question. 
To do with ` dut.TEST_PARAMETERS_FILENAME` would require some changes in cocotb itself.

benreynwar, 2019-10-23:
Presumably this would have issues when I try to run multiple simulations in parallel.
But it's a good solution for now.  Thanks!

closed 2019-10-23 by benreynwar

themperek, 2019-10-23:
It should work for parallel runs.

Something like this with works with `pytest-xdist` in parallel.
```python
@pytest.mark.parametrize('seed', range(1000))
def test_basic(seed):
    extra_env = {"RESULTS_FILE_NAME", "abc_"+str(seed)+".txt")}
    run(
        seed=seed,
        extra_env=extra_env
    )
```

benreynwar, 2019-10-23:
Oh yeah, I guess since it's spawning separate processes they'll all have independent environment variables.  I was brain farting.