themperek/cocotb-test #59: Python 2.7 regex problem

garmin-mjames, 2019-12-09:
`setup.py` lists Python 2.7 support, but `re.ASCII` is not a valid flag in Python 2.
https://github.com/themperek/cocotb-test/blob/1dc351d971254d10811b589028322c35f67ac9cd/cocotb_test/simulator.py#L11

themperek, 2019-12-12:
How does this pass CI? https://dev.azure.com/themperek/themperek/_build/results?buildId=384

garmin-mjames, 2019-12-16:
The `Linux-Icarus-py27` build shows `pytest` running on `Python 3.7.4`:

https://dev.azure.com/themperek/themperek/_build/results?buildId=384&view=logs&j=46cbf7f6-cf03-5b83-7bb6-f9f6586217d5&t=9d2e9dca-ef2c-54ea-2e89-912dd36b5a60&l=13

Could be a configuration issue.

themperek, 2020-01-03:
@garmin-mjames Thank you for reporting!

It should be fixed now also CI should now be correct.

closed 2020-01-03 by themperek