themperek/cocotb-test #60: run_filename is not what you want when running in virtualenv

markmelvin, 2019-12-31:
I noticed that when I run pytest with pipenv (which is fairly standard), the `run_filename` passed to the simulator is the parent directory of the Python in the virtual environment (e.g. `/home/<user>/.local/share/virtualenvs/<foldername>-ffNzxzuc/lib/python3.7/site-packages/_pytest/python.py`). Since this is later used to resolve relative verilog sources, includes, etc. it is probably not a good idea to use the inspect module to figure out the path of the running file.

themperek, 2020-01-01:
Yes it is here: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/run.py#L20
I do not like this too but did not have a better idea for now how to do this for now.

themperek, 2022-02-18:
This has changed so closing it.
In case still an issue please reopen.