themperek/cocotb-test #79: ghdl with latest release

bth105, 2020-05-11:
Hi i have just updated my test enviroment to the latest version of cocotb-test. In the requirements of cocotb-test is a dependancy on cocotb 1.3.1. which as far as i can tell has a issue with GHDL where i get the error "Couldn't find makefile for simulator: "ghdl"! Available simulators: " - issue #1699 in cocotb. Is there any way we can change the dependency to 1,4.0 dev as i know that works

themperek, 2020-05-11:
This is probably not cocotb 1.3.1 issue. You probably use examples/documentation (makefile) for development 1.4 and not 1.3.1.

Please let me know.
Will be updating cocotb-test for 1.4 soon.

bth105, 2020-05-11:
Hi thanks for getting back to me.
I have only changed the 
include $(shell cocotb-config --makefiles)/Makefile.sim
in the makefile since i first started using cocotb v1.1. So if 1.3.1 has something different for that version then maybe. I have also found some difference in the way the callback passes data in avalon pkt monitors which is specific to 1.3.1 as i have the tests working with 1.4.0 dev. I may wait for the update of cocotb-test to 1.4.

themperek, 2020-05-11:
Do you have `include $(shell cocotb-config --makefiles)/Makefile.inc` ?

bth105, 2020-05-12:
I hadn't but i have tried it with that and i still get the same error. But dont worry to much as i know this project doesn't use the makefiles

themperek, 2020-07-09:
I will close this. Please feel free to reopen if needed.

closed 2020-07-09 by themperek