themperek/cocotb-test #90: The cocotb logs only show up in the terminal, and not dumped to file.

gretzteam, 2020-07-20:
This is with xcelium, but potentially applies to other simulator.
All the log messages from cocotb, such as:
`self.dut._log.info("Doing something")`
do not appear in the compilation/simulation logs.

gretzteam, 2020-07-20:
I looked at the code in the execute() command of the Simulator Class:

If I replace the following code:

```
            self.process = subprocess.Popen(
                cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT, cwd=self.work_dir, env=self.env
            )

            while True:
                out = self.process.stdout.readline()

                if not out and self.process.poll() is not None:
                    break

                log_out = out.decode("utf-8").rstrip()
                if log_out != "":
                    self.logger.info(log_out)

            if self.process.returncode:
                self.logger.error("Command terminated with error %d" % self.process.returncode)
                return
```
with somethign like this, we get a cocolog.log file in the work directory (stripped of the ANSI character that make the log hard to read on VS code). 

```
            with subprocess.Popen(cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT, cwd=self.work_dir, env=self.env, bufsize=1) as p, \
                open(self.work_dir+'/cocolog.log', 'w') as file:
                for line in p.stdout: 
                    sys.stdout.buffer.write(line) 
                    escaped_line = self.escape_ansi(line.decode('utf-8'))
                    file.write(escaped_line)

            if p.returncode:
                self.logger.error("Command terminated with error %d" % p.returncode)
                return
```
This needs to be defined in the class as well:

```
    def escape_ansi(self, line):
        ansi_escape = re.compile(r'(\x9B|\x1B\[)[0-?]*[ -/]*[@-~]')
        return ansi_escape.sub('', line)
```

I'm not familiar at all with how subprocess works, and most of this is copy paste from here:
https://stackoverflow.com/questions/15535240/python-popen-write-to-stdout-and-log-file-simultaneously

In any case it could be a starting point.

themperek, 2022-02-18:
This has changed quite a bit.  I am closing it.
In case still an issue please reopen.

closed 2022-02-18 by themperek