themperek/cocotb-test #92: Missing documentation

sfavazza-duagon, 2020-08-13:
Hello there, I'm using cocotb for quite a while now and I recently discovered this package. I can finally get rid of the makefiles!
Yet I was wondering whether there is any plan to document the methods defined by this module.

I went through the code, despite it is often clear, there are some `kwargs` that it is not so trivial to figure out what their purpose is. Perhaps it would be handy to have a quick look to documentation to make sense of everything available to the user.

Let me know if I can help in this task.

cmarqu, 2020-08-13:
The idea is to eventually have the functionality of this package in cocotb proper, so some of the docstring work would be gone to waste - on the other hand, it may take a while until that happens, so there also is value to it...
A pull request would certainly quickly be merged, at least if it's only documentation.

sfavazza-duagon, 2020-08-14:
I understand your point, and yes, during this transition some documentation it would be nice.

> so some of the docstring work would be gone to waste

I don't get why, do you plan to suppress some functions here and there when integrating it in cocotb?

I am currently using it in my project with modelsim, but I had to fix something as in the current state it would not work (I had paths and command generation issues).
Even if off-topic, would it be possible to share the Travis test environment already setup in cocotb? It would be nice to contribute and be sure to not break anything...
On my side I can only test locally in modelsim. If a test environment with all other supported simulators would be available I could also ask for a bug-fix merge request. I want to stress this point as I could not make cocotb-test work with the current usage-guide.

themperek, 2020-12-30:
I have added minimal documentation in `README.md`
Contributions welcome.

closed 2020-12-30 by themperek