themperek/cocotb-test #94: Allow file specific compile arguments

marph91, 2020-09-08:
As far as I understood, there is no possibility to have file specific compile arguments. My source files should be compiled into different libraries, like:

```bash
ghdl -a --work=lib_foo file1.vhd
ghdl -a --work=lib_bar file2.vhd
```

At first, is my assumption correct?
Do you think it would be suitable to add another argument (for example `file_specific_compile_args`) to the `Simulator` class, which must have the same length as `vhdl_sources`, respectively `verilog_sources`?

themperek, 2020-09-08:
At the moment one would need to customize compiler similar like here: https://github.com/themperek/cocotb-test/blob/master/tests/test_dff_custom_sim.py#L8-L10

Add overload `build_command` function (add custom compile sequnce).

Not sure what would be best:
- stay like is
- add a way to register pre/post functions/commands
- make a plugin for [edalize](https://github.com/olofk/edalize ) and let it handle this 
- ?

One can also can try to make ghdl support a bit better like Questa: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/simulator.py#L351-L421
- add `rtl_library` parameter (`--work=rtl_library`)
- call with `compile_only` with specific `rtl_library`
- call `run`

Not sure ...

marph91, 2020-09-27:
For now, I [customized the compiler ](https://github.com/marph91/pico-png/blob/cocotb-poc/sim/png_encoder/test_png_encoder.py#L20-L46) and adapted it to my needs.

closed 2020-09-27 by marph91