themperek/cocotb-test #95: Cannot specify a command file for Icarus from compile_args - Needed for setting timescale

gretzteam, 2020-09-12:
To set the icarus timescale, a cmds.f file with the following content is created:

`+timescale+1ns/1ns
`

Then compile_args is set to:
compile_args = ["-f /home/cmds.f"]

However, icarus complains that the file cannot be read:

```
INFO cocotb: Running command: iverilog -o /home/sim_build/toplevel.vvp -D COCOTB_SIM=1 -s toplevel -g2012 -f /home/cmds.f  /home/toplevel.sv 
INFO cocotb: iverilog: cannot open command file  /home/cmds.f for reading.
ERROR cocotb: Command terminated with error 1
AssertionError: Simulation terminated abnormally. Results file not found.
```

Interestingly, if I then manually go in /sim_build and type the exact same command, there are no errors and subsequent run work fine...
`iverilog -o /home/sim_build/toplevel.vvp -D COCOTB_SIM=1 -s toplevel -g2012 -f /home/cmds.f  /home/toplevel.sv 
`

themperek, 2020-09-14:
Try :
`compile_args = ["-f", "/home/cmds.f"]`

Should probably split this in code before passing to subprocess.

gretzteam, 2020-09-14:
This worked fine, thanks!

Related to this, would it make sense to add a default cmds.f file with decent timescale directive? I believe Icarus is mostly not usable as it is for most people and they will all have to figure out the funny way to set the timescale...

Something along those lines, but I'm sure there is a better way.

```
class icarus_custom(Icarus):
    def compile_command(self):
        if '-f' in self.compile_args:
            print('A command file is already provided.')
        else:
            print('Override default icarus timescale to 1ns/1ns. Create cmds.f file and add to compile_args')
            with open(self.sim_dir+"/cmds.f", 'w') as f:
                f.write("+timescale+1ns/1ns")  
            self.compile_args.extend(["-f", self.sim_dir+"/cmds.f"])

        cmd_compile = (
            ["iverilog", "-o", self.sim_file, "-D", "COCOTB_SIM=1", "-s", self.toplevel, "-g2012"]
            + self.get_define_commands(self.defines)
            + self.get_include_commands(self.includes)
            + self.compile_args
            + self.verilog_sources
        )

        return cmd_compile
```

themperek, 2022-02-18:
I am closing it.
In case it is still an issue please reopen.

closed 2022-02-18 by themperek

shuckc, 2022-11-19:
This is one of the things I imagine every Icarus user has to solve when moving from `cocotb` to `cocotb-test` - why is Icarus now complaining about timescales?

The Cocotb Makefile takes care of writing a default timescale to `cmds.f` file and including it here: https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.icarus#L69
When cocotb-test invokes icarus it has no equivalent to set timescale https://github.com/themperek/cocotb-test/blob/master/cocotb_test/simulator.py#L421 

Setting `compile_args` on every test is clunky, and it's not obvious how to replace `SIM=icarus pytest .` with a custom `Icarus` subclass (like `icarus_custom` above) without losing the ability to switch simulators from the command line.