themperek/cocotb-test #99: Cannot set verilator 'timescale'

gretzteam, 2020-09-20:
From the [cocotb doc](https://docs.cocotb.org/en/latest/simulator_support.html#verilator), Verilator handles timescale a bit differently than other simulators.

Looking at the verilator [Makefile](https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.verilator)
we ultimately need to append the -DVL_TIME_PRECISION_STR switch to the CPPFLAGS env var.

```
ifdef COCOTB_HDL_TIMEPRECISION
  SIM_BUILD_FLAGS += -DVL_TIME_PRECISION_STR=$(COCOTB_HDL_TIMEPRECISION)
endif
SIM_BUILD_FLAGS += -std=c++11
# Compilation phase
$(SIM_BUILD)/$(TOPLEVEL): $(SIM_BUILD)/Vtop.mk
	CPPFLAGS="$(SIM_BUILD_FLAGS)" make -C $(SIM_BUILD) -f Vtop.mk
```

In cocotb=test, the Verilator class doesn't allow for adding more options to CPPFLAGS

`        self.env["CPPFLAGS"] = "-std=c++11"
`       

I'm not sure what is the best way to support this while keeping things consistent with other simulators.

themperek, 2020-09-21:
I think Verilator has timescale support now. I will chek how to support this.

gretzteam, 2020-09-21:
Independently, should this line:
```
self.env["CPPFLAGS"] = "-std=c++11"
```
be moved to the __init__() of the verilator class? That way we can append to it using a verilator_custom class. Right now it's impossible to add custom CPPFLAGS.

themperek, 2020-09-21:
> Independently, should this line:
> 
> ```
> self.env["CPPFLAGS"] = "-std=c++11"
> ```
> 
> be moved to the **init**() of the verilator class?

YES

themperek, 2020-10-15:
@gretzteam Would this be enough (can add via `extra_env)? https://github.com/themperek/cocotb-test/pull/104/files 

`.

themperek, 2020-10-19:
It should be easier now (https://github.com/themperek/cocotb-test/pull/104). In case you have other/better ideas please reopen.

closed 2020-10-19 by themperek