themperek/cocotb-test #100: GHDL waveform

qarlosalberto, 2020-10-06:
Hi,

I'm having problems to generate the waveform with cocotb-test and GHDL:
- GHDL:
```
GHDL 0.36 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 7.4.0
 GCC back-end code generator
```
- pytest (5.3.5)
- cocotb-test (0.1.0)
- cocotb (1.4.0)

I can generate the .vcd waveform with a simple Makefile:

```

TOPLEVEL_LANG=vhdl
VHDL_SOURCES =  $(PWD)/../../RTL_code/sample.vhd
TOPLEVEL=sample
MODULE=sample_cocotb
SIM=ghdl
COMPILE_ARGS=--ieee=synopsys
SIM_ARGS=--vcd=func.vcd
include $(shell cocotb-config --makefiles)/Makefile.inc
include $(shell cocotb-config --makefiles)/Makefile.sim
```

But no with cocotb-test:
```
from cocotb_test.simulator import run
import os

CURRENT_PATH = os.path.dirname(os.path.abspath(__file__)) + '/'

def test_modulator():
    run(
        vhdl_sources=[CURRENT_PATH +
                      "/../../RTL_code/sample.vhd"],
        toplevel="sample",
        module="sample_cocotb",
        toplevel_lang="vhdl",
        sim_args=["--vcd=func.vcd"],
        compile_args=["--ieee=synopsys"]
    )
```

marph91, 2020-10-10:
It works for me with GHDL 1.0-dev (v0.37.0-959-ga18e1503). The waveform is located at the `sim_build` folder.

qarlosalberto, 2020-10-13:
Is there a way to see the GHDL command? A debug output?

I'm using the master version of cocotb-test

marph91, 2020-10-14:
I'm not aware of any debug output.

In your case it might be a naming issue. The `Simulator` class expects an parameter with the name `simulation_args` instead of `sim_args`. See https://github.com/themperek/cocotb-test/blob/master/cocotb_test/simulator.py#L43.

qarlosalberto, 2020-10-14:
Ohhh yes, the problem was in ```simulation_args```

Thank you! :)

themperek, 2020-10-15:
Should we change to `simulation_args` to `sim_args`?

qarlosalberto, 2020-10-15:
I think yes, it's a little confused. And it doesn't show any error, you don't know what is happen :/

qarlosalberto, 2020-10-15:
Other idea is to show an error if you pass an invalid argument:

    run(
        verilog_sources=[os.path.join(example_dir, "ping_tun_tap", "hdl", "icmp_reply.sv")],
        toplevel="icmp_reply",
        python_search=[os.path.join(example_dir, "ping_tun_tap", "tests")],
        module="test_icmp_reply",
        **bad_argument**="other"
    )

themperek, 2020-10-15:
>Other idea is to show an error/warning if you pass an invalid argument:

This would make customization more painful. But maybe makes sense to change this.
Remove this line: https://github.com/themperek/cocotb-test/blob/f67762202aa0c48b4db8488a07b8010836c96694/cocotb_test/simulator.py#L53

PR #112 (2020-10-26): Deprecate simulation_args and kwargs
Closes #100
closed 2020-10-26 by themperek