themperek/cocotb-test #107: Support for waveform dumps

alexforencich, 2020-10-18:
Support for enabling waveform dumping should be added. Different simulators need module parameters passed through in different ways - different command line switches are used, and some require them in `SIM_ARGS`, some in `COMPILE_ARGS`.

Verilator accepts `--trace` and hopefully soon `--trace-fst` in `COMPILE_ARGS`. Icarus Verilog requires a little bit more work, the best option I think is to write out a module that contains `$dumpfile("toplevel.fst");` and `$dumpvars(0, toplevel)`, add this to the sources, and add a `-s mod_name` to `COMPILE_ARGS` to select it as an additional top-level module.  It's also necessary to select the dump file format in plusargs (e.g. `-fst`).  No idea about other simulators, although it seems some of them can look for a `WAVES` env var.  

Having a standardized way to support this in cocotb-test would be very convenient, perhaps via an argument to `Simulator.__init__()`.

cmarqu, 2020-10-19:
While developing the UI for this, we might want to support features like "allowlist" and "denylist" (maybe even regexes) for signal names - maybe not initially, but with such things in mind.

alexforencich, 2020-10-19:
I'm thinking maybe the best way forward is to support the lowest common denominator - in other words, only things that work across all simulators, or at least the majority of commonly-used simulators.  At least for verilator, it seems to be either on or off.  Icarus verilog is little better - either you dump everything, or you can pick a few signals to dump.  Not sure about the other simulators.  Providing a simple "dump all the things" option, while leaving more complex, simulator-specific configuration to be done manually I think may be a reasonable compromise.

alexforencich, 2020-11-29:
See #116