themperek/cocotb-test #122: unrecognized arguments: --cov=cocotb --cov-branch

catkira, 2021-01-01:
When I create the test_dff.py as described in readme and run pytest I get the following error:

> 
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --cov=cocotb --cov-branch
  inifile: /mnt/d/git/cocotb/setup.cfg
  rootdir: /mnt/d/git/cocotb

I am using python 3.8, Ubunt 18, cocotb and cocotb-test from pip

themperek, 2021-01-01:
You probably need `pip3 install pytest-cov`

But where is this used in `cocotb-test`?

cmarqu, 2021-01-02:
Usage of coverage is probably configured in a pytest config file.

themperek, 2021-01-02:
Yes but not in `cocotb-test`. I cannot see this used anywhere here?

catkira, 2021-01-02:
When I do pytest --collect-only I get this
root@LAPTOP-Benjamin:/mnt/d/git/cocotb/examples/dff# pytest --colect-only
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: --colect-only
  inifile: /mnt/d/git/cocotb/setup.cfg
  rootdir: /mnt/d/git/cocotb

root@LAPTOP-Benjamin:/mnt/d/git/cocotb/examples/dff# pytest --collect-only
================================================= test session starts ==================================================
platform linux -- Python 3.8.0, pytest-6.2.1, py-1.10.0, pluggy-0.13.1 -- /usr/bin/python3
cachedir: .pytest_cache
rootdir: /mnt/d/git/cocotb, configfile: setup.cfg
plugins: cocotb-test-0.1.1, cov-2.10.1
collected 1 item

<Module examples/dff/test_dff.py>
  <Function test_dff>

----------- coverage: platform linux, python 3.8.0-final-0 -----------
Name                                                                   Stmts   Miss Branch BrPart  Cover
--------------------------------------------------------------------------------------------------------
/usr/local/lib/python3.8/dist-packages/cocotb/ANSI.py                     49      0      0      0   100%
/usr/local/lib/python3.8/dist-packages/cocotb/__init__.py                126     93     42      0    20%
/usr/local/lib/python3.8/dist-packages/cocotb/_os_compat.py                7      2      2      1    67%
/usr/local/lib/python3.8/dist-packages/cocotb/_py_compat.py               13      5      2      1    60%
/usr/local/lib/python3.8/dist-packages/cocotb/_version.py                  1      0      0      0   100%
/usr/local/lib/python3.8/dist-packages/cocotb/binary.py                  441    324    174      1    19%
/usr/local/lib/python3.8/dist-packages/cocotb/bus.py                      69     59     34      0    10%
/usr/local/lib/python3.8/dist-packages/cocotb/clock.py                    39     26      8      0    28%

I dont understand why it analyzes the coverage of my python. After installing pytest-cov it runs a bit further.

root@LAPTOP-Benjamin:/mnt/d/git/cocotb/examples/dff/hdl# pytest -o log_cli=True
================================================= test session starts ==================================================
platform linux -- Python 3.8.0, pytest-6.2.1, py-1.10.0, pluggy-0.13.1 -- /usr/bin/python3
cachedir: .pytest_cache
rootdir: /mnt/d/git/cocotb, configfile: setup.cfg
plugins: cocotb-test-0.1.1, cov-2.10.1
collected 1 item

test_dff.py::test_dff
---------------------------------------------------- live log call -----------------------------------------------------
WARNING  cocotb:simulator.py:357 Skipping compilation:/mnt/d/git/cocotb/examples/dff/hdl/sim_build/dff.vvp
INFO     cocotb:simulator.py:244 Running command: vvp -M /usr/local/lib/python3.8/dist-packages/cocotb/libs -m libcocotbvpi_icarus /mnt/d/git/cocotb/examples/dff/hdl/sim_build/dff.vvp
INFO     cocotb:simulator.py:255      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
INFO     cocotb:simulator.py:255      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
INFO     cocotb:simulator.py:255      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Icarus Verilog version 12.0 (devel)
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb                                      __init__.py:208  in _initialise_testbench           Running tests with cocotb v1.4.0 from /usr/local/lib/python3.8/dist-packages/cocotb
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb                                      __init__.py:229  in _initialise_testbench           Seeding Python random module with 1609584890
INFO     cocotb:simulator.py:255      0.00ns CRITICAL cocotb.regression                         regression.py:177  in _discover_tests                 Failed to import module dff_cocotb: No module named 'dff_cocotb'
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb.regression                         regression.py:178  in _discover_tests                 MODULE variable was "dff_cocotb"
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb.regression                         regression.py:179  in _discover_tests                 Traceback:
INFO     cocotb:simulator.py:255      0.00ns INFO     cocotb.regression                         regression.py:180  in _discover_tests                 Traceback (most recent call last):
INFO     cocotb:simulator.py:255
                                File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 175, in _discover_tests
INFO     cocotb:simulator.py:255
                                  module = _my_import(module_name)
INFO     cocotb:simulator.py:255
                                File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 69, in _my_import
INFO     cocotb:simulator.py:255
                                  mod = __import__(name)
INFO     cocotb:simulator.py:255
                              ModuleNotFoundError: No module named 'dff_cocotb'
INFO     cocotb:simulator.py:255
[... cut ...]
INFO     cocotb:simulator.py:255      0.00ns ERROR    cocotb.scheduler                            __init__.py:269  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
FAILED                                                                                                           [100%]^C
INTERNALERROR> Traceback (most recent call last):
INTERNALERROR> AttributeError: 'Icarus' object has no attribute 'process'

I used the test_dff.py from the README file and copied it into the hdl folder of the cocotb dff example.

themperek, 2021-01-02:
- Because you run in cocotb folder and it takes configuration form: 
https://github.com/cocotb/cocotb/blob/2e392918c84e6222922e15ef59d28f5abdc3cdb7/setup.cfg#L32

- `INFO cocotb:simulator.py:255 0.00ns CRITICAL cocotb.regression regression.py:177 in _discover_tests Failed to import module dff_cocotb: No module named 'dff_cocotb'`
You need to have `dff_cocotb` module is same directory or add location  `python_search`?

catkira, 2021-01-02:
is the dff_cocotb module somewhere or do I have to create it myself?

themperek, 2021-01-02:
For example here: https://github.com/themperek/cocotb-test/blob/master/tests/dff_cocotb.py and you need https://github.com/themperek/cocotb-test/blob/master/tests/dff.v also.

catkira, 2021-01-02:
When I run it in the tests dir of cocotb-test I get:
root@LAPTOP-Benjamin:/mnt/d/git/cocotb-test# pytest -o log_cli=True
================================================= test session starts ==================================================
platform linux -- Python 3.8.0, pytest-6.2.1, py-1.10.0, pluggy-0.13.1
rootdir: /mnt/d/git/cocotb-test
plugins: cocotb-test-0.1.1, cov-2.10.1
collected 30 items / 2 errors / 28 selected

======================================================== ERRORS ========================================================
____________________________________ ERROR collecting tests/test_cocotb_examples.py ____________________________________
tests/test_cocotb_examples.py:10: in <module>
    raise IOError("Cocotb example directory not found. Please clone with git and install with `pip -e`")
E   OSError: Cocotb example directory not found. Please clone with git and install with `pip -e`
_____________________________________ ERROR collecting tests/test_cocotb_tests.py ______________________________________
tests/test_cocotb_tests.py:11: in <module>
    raise IOError(
E   OSError: Cocotb test directory not found. Please clone with git and install with `pip -e`
=================================================== warnings summary ===================================================
tests/test_parallel.py:15
  /mnt/d/git/cocotb-test/tests/test_parallel.py:15: PytestUnknownMarkWarning: Unknown pytest.mark.compile - is this a typo?  You can register custom marks to avoid this warning - for details, see https://docs.pytest.org/en/stable/mark.html
    @pytest.mark.compile

../../../../usr/local/lib/python3.8/dist-packages/cocotb/result.py:125
  /usr/local/lib/python3.8/dist-packages/cocotb/result.py:125: PytestCollectionWarning: cannot collect test class 'TestFailure' because it has a __init__ constructor (from: tests/test_parameters.py)
    class TestFailure(TestComplete, AssertionError):

../../../../usr/local/lib/python3.8/dist-packages/cocotb/result.py:125
  /usr/local/lib/python3.8/dist-packages/cocotb/result.py:125: PytestCollectionWarning: cannot collect test class 'TestFailure' because it has a __init__ constructor (from: tests/test_plus_args.py)
    class TestFailure(TestComplete, AssertionError):

-- Docs: https://docs.pytest.org/en/stable/warnings.html
=============================================== short test summary info ================================================
ERROR tests/test_cocotb_examples.py - OSError: Cocotb example directory not found. Please clone with git and install ...
ERROR tests/test_cocotb_tests.py - OSError: Cocotb test directory not found. Please clone with git and install with `...
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================ 3 warnings, 2 errors in 0.51s =============================================
root@LAPTOP-Benjamin:/mnt/d/git/cocotb-test#

is it really necessary that I install cocotb with -e ?

themperek, 2021-01-02:
>is it really necessary that I install cocotb with -e ?

Yes if you want to run some test/examples from cocotb. In all other cases no.

catkira, 2021-01-02:
ok, I guess because of this line:
example_dir = os.path.join(os.path.dirname(os.path.dirname(cocotb.__file__)), "examples")

I changed it to 
example_dir = "/mnt/d/git/cocotb/examples/"

the result looks ok (I dont have vhdl sims installed because I dont use it)
FAILED test_cocotb_examples.py::test_ping_tun_tap - AssertionError: None class="test_icmp_reply" test="tun_tap_exampl...
FAILED test_cocotb_examples.py::test_dff_vhdl - ValueError: This simulator does not support VHDL
FAILED test_cocotb_examples.py::test_adder_vhdl - ValueError: This simulator does not support VHDL
FAILED test_cocotb_examples.py::test_mean - ValueError: This simulator does not support VHDL
FAILED test_cocotb_examples.py::test_mixed_top_verilog - ValueError: This simulator does not support VHDL
FAILED test_cocotb_examples.py::test_mixed_top_vhdl - ValueError: This simulator does not support VHDL
FAILED test_cocotb_examples.py::test_endian_swapper_vhdl - ValueError: This simulator does not support VHDL
FAILED test_cocotb_tests.py::test_verilog_access - NameError: name 'tests_dir' is not defined
FAILED test_dff.py::test_dff_vhdl - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[0] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[1] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[2] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[3] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[4] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[5] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[6] - ValueError: This simulator does not support VHDL
FAILED test_parallel.py::test_dff_vhdl_param[7] - ValueError: This simulator does not support VHDL
FAILED test_parameters.py::test_dff_vhdl_testcase[parameters0] - ValueError: This simulator does not support VHDL
FAILED test_parameters.py::test_dff_vhdl_testcase[parameters1] - ValueError: This simulator does not support VHDL
===================== 19 failed, 18 passed, 2 skipped, 2 xfailed, 1 xpassed, 31 warnings in 35.93s =====================

Thank for the help!

themperek, 2021-01-02:
So this is for cocotb-test testing.
You would need to set `SIM=icarus` so it will not try to run VHDL examples. 

I will do it a bit smarter.
Should I:
1.  do not an error but skip the test in this case 
2. assume Icarus when not `SIM` set
?

No. 2 looks safer.

catkira, 2021-01-02:
I would not assume icarus, because at some point more people could be using verilator. Is it possible to separately set a sim for verilog and vhdl?

I would add in the README where to put the test_dff.py file, or mention that it already exists in the repo.
To rely on the symbolic links for finding the cocotb dir is a bit ugly. I would prefer to require the user to set a environment variable to run the cocotb tests, ie COCOTBDIR = \mnt\d\..., but I would put some examples in the repo that run out of the box and just disable the cocotb tests if COCOTDIR is not set.
Because if you are not a cocotb dev, you just want to have the normal release of cocotb installed.

themperek, 2021-01-02:
I have changed the README. Is this better now?

catkira, 2021-01-02:
The final result shows  
2 xfailed, 1 xpassed
What does xfailed and xpassed mean?

catkira, 2021-01-02:
> I have changed the README. Is this better now?

yes, that could prevent some questions in the future

themperek, 2021-01-02:
> The final result shows
> 2 xfailed, 1 xpassed
> What does xfailed and xpassed mean?

That those tests are expected to fail.
https://docs.pytest.org/en/latest/skipping.html#xfail-mark-test-functions-as-expected-to-fail

catkira, 2021-01-02:
Its a bit off topic in this issue, but anyway:
I would also add to the readme or a separate documentation what cocotb-test is and what the main advantages are. It was not really clear to me until I wanted to test verilog modules for different parameters. That was not possible with just cocotb because it only compiles the sources for one set of fixed parameters. (I found out about cocotb-test by looking how alexforencich
implemented the tests for different parameters). Maybe there are other advantages, but that was the main point for me.

themperek, 2021-01-02:
> Its a bit off topic in this issue, but anyway:
> I would also add to the readme or a separate documentation what cocotb-test is and what the main advantages are. It was not really clear to me until I [...]

Can add somewhere but this is not cocotb-test just standard `pytest`. Same as running in parallel etc.

catkira, 2021-01-02:
> > The final result shows
> > 2 xfailed, 1 xpassed
> > What does xfailed and xpassed mean?
> 
> That those tests are expected to fail.
> https://docs.pytest.org/en/latest/skipping.html#xfail-mark-test-functions-as-expected-to-fail

ah ok, so I can mark them mentally as green. I would also print them in green, but ok its how pytest does it.

catkira, 2021-01-02:
> Can add somewhere but this is not cocotb-test just standard `pytest`. Same as running in parallel etc.

In the cocotb git it says somewhere that you cannot change parameters dynamically for testing, you have to pass compile args to set them. After reading that I was disappointed and only relieved after I found cocotb-test. I think cocotb-test should be the default and included in cocotb. It makes it much more useful.
I am writing an axi stream component that reacts differently if the input and output data width match or not match. With only cocotb I cannot really test it completely.

themperek, 2021-01-02:
>I am writing an axi stream component that reacts differently if the input and output data width match or not match. With only cocotb I cannot really test it completely.

Good luck.
Let me know us know if you have any troubles,

Can this be closed?

closed 2021-01-02 by catkira