themperek/cocotb-test #125: Accept simulator as a parameter to cocotb_test.simulator.run

alexforencich, 2021-01-09:
It would be nice if the simulator could be selected without setting an env var.  Ideally, `cocotb_test.simulator.run` should be updated to accept a `simulator` argument that could select the simulator.  Not sure what precendence would make sense with respect to the `SIM` env var.

jonpovey, 2021-12-03:
This would be nice for my use. Could allow parameterising a test across multiple simulators for regressions, if you wanted that sort of thing. I just want it because I'm always on the same simulator.
Also not sure about precedence. Maybe it should just error out if the simulator is set by both mechanisms and they clash.

It would also not cooperate easily with the examples that use the mechanism
`@pytest.mark.skipif(os.getenv("SIM") == "ghdl", reason="Verilog not suported")`

themperek, 2021-12-03:
One can add `simulator` argument here (take priority over `SIM` environmental variable) : https://github.com/themperek/cocotb-test/blob/19b1cd087407926005ac8c865934455e913b8b76/cocotb_test/simulator.py#L1021

themperek, 2022-02-18:
This should be in https://github.com/themperek/cocotb-test/pull/167

closed 2022-02-18 by themperek