themperek/cocotb-test #126: SIM_BUILD var not found at run time

aignacio, 2021-02-12:
Hey,
I'm trying to run the following code below using Cocotb-test and when I run through the makefile, it finishes without no issues, but when I launch with pytest, it complains about the not finding the env SIM_BUILD var that's dynamically build for each case of test.

https://github.com/aignacio/async_gp_fifo/blob/master/test_fifo_async.py#L33

```bash
INFO     cocotb:simulator.py:255     40.00ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: run_test_001 (result was KeyError)
INFO     cocotb:simulator.py:255                                                                                                                      Traceback (most recent call last):
INFO     cocotb:simulator.py:255                                                                                                                        File "/usr/local/lib/python3.9/site-packages/cocotb/regression.py", line 599, in _my_test
INFO     cocotb:simulator.py:255                                                                                                                          await function(dut, *args, **kwargs)
INFO     cocotb:simulator.py:255                                                                                                                        File "/Users/aignacio/projects/asyng_gp_fifo/test_fifo_async.py", line 115, in run_test
INFO     cocotb:simulator.py:255                                                                                                                          ff_driver = AFIFODriver(signals=dut,debug=True)
INFO     cocotb:simulator.py:255                                                                                                                        File "/Users/aignacio/projects/asyng_gp_fifo/test_fifo_async.py", line 33, in __init__
INFO     cocotb:simulator.py:255                                                                                                                          sim_build     = os.environ['SIM_BUILD']
INFO     cocotb:simulator.py:255                                                                                                                        File "/usr/local/Cellar/python@3.9/3.9.1_8/Frameworks/Python.framework/Versions/3.9/lib/pytho
n3.9/os.py", line 679, in __getitem__
INFO     cocotb:simulator.py:255                                                                                                                          raise KeyError(key) from None
INFO     cocotb:simulator.py:255                                                                                                                      KeyError: 'SIM_BUILD'
```

What might be the case to check before launching the test in addition of SIM_NAME?

themperek, 2021-02-13:
`SIM_BUILD` is not set as an environmental variable. Maybe should be?
In `cocotb-test` most of time you can do `os.getcwd()`

The log file is forwarded from cocotb/simulator to top-level/control process so you could log it there (this would also log simulator messages). In general, logging needs a bit of thinking still.

aignacio, 2021-02-13:
Thanks @themperek but I've tried to set with ** extra_env** and the behaviour is the same, I just realize that the path dir at runtime is sim_build so I remove all prepends from the path and it was inside the sim folder I wanna since the beginning. Just a small question about logging, I saw that there're a lot of duplicated messages in my log, is it a side effect of using two clock domains in the simulation or you never saw something similar?

closed 2021-02-13 by aignacio

themperek, 2021-02-13:
> Just a small question about logging, I saw that there're a lot of duplicated messages in my log, is it a side effect of using two clock domains in the simulation or you never saw something similar?

No idea. 

You should be able to do something like this also (use `RotatingFileHandler` from top not inside cocotb):

```python
@pytest.mark.parametrize("slots",[2,4,8,16,32,64,128])
def test_fifo_async(slots):

    sim_build = os.path.join(tests_dir, "sim_build_afifo_"+"_".join(("{}={}".format(*i) for i in parameters.items())))
    os.makedirs(sim_build, exist_ok=True)

    logger = logging.getLogger()
    handler = RotatingFileHandler(sim_build+"/sim.log")
    logger.addHandler(handler)

    run(
        ...
        sim_build=sim_build,
    )

    logger.removeHandler(handler)
```