themperek/cocotb-test #127: Argument on Xcelium build forces pre-step setup

aignacio, 2021-02-20:
Hi,

could you please remove the default argument "-elaborate" on Xcelium/Ius simulator, to use that you need to force work lib creation previously, what doesn't happen at time 0 when you're running the sims and by default xrun will know what to call first depending the stage you're. I was wondering why I couldn't run the same tests with cocotb-test that I successfully run with standard makefile flow and removing this argument made it work.

https://github.com/themperek/cocotb-test/blob/3ed5aac505dbe85f589413be7d14eaee3458d781/cocotb_test/simulator.py#L607

themperek, 2021-02-23:
As far as I understand I have split the run on elaboration/compilation and execution.  I still would like to use `compile_only`. Not sure why it would matter?

> I was wondering why I couldn't run the same tests with cocotb-test that I successfully run with standard makefile flow and removing this argument made it work.

This is I do not understand? Why if would not work?

aignacio, 2021-02-23:
Let me re-write that sentence, the same test that works fine with standard makefile flow (without Cocotb-test) on Xcelium/Ius doesn't work when running through cocotb-test/pytest bc of this issue with the simulator:
```bash
INFO     cocotb:simulator.py:255 define XMSIMRC	( /mnt/simvision/tools/xcelium/files/xmsimrc, ~/.xmsimrc )
INFO     cocotb:simulator.py:255 |
INFO     cocotb:simulator.py:255 xrun: *F,DLHVAR (./xcelium.d/run.lnx8664.18.03.d/hdl.var,1): hdl.var invalid variable '~ == $HOME'.
ERROR    cocotb:simulator.py:258 Command terminated with error 2
```

themperek, 2021-02-24:
Did you run in the same directory `make` and `pytest`?
Did you clean `make clean` or `cocotb-clean` between the runs?

I cannot reproduce this.

aignacio, 2021-02-24:
Hey @themperek, I followed these steps but I'm still seeing the issues, I'm attaching the two logs, one with the run through tox (Cocotb-test) and another one with Makefile, I'm attaching the sim_build folders, obviously the one with tox is considering just one of the parameters in the test but it works pretty much in the same manner. Would you mind to take a look and let me know that's the difference between the two executions?

[makefile_run.zip](https://github.com/themperek/cocotb-test/files/6038850/makefile_run.zip)
[tox_run.zip](https://github.com/themperek/cocotb-test/files/6038852/tox_run.zip)

themperek, 2021-02-25:
Is the same happening on any example? Like `tests/test_dff.py` ? This make very little sense.

themperek, 2021-02-25:
You can also `easily` customize the simulator:

https://github.com/themperek/cocotb-test/blob/master/tests/test_dff_custom_sim.py (I have to improve this example).

aignacio, 2021-02-25:
I had the same error with tests/test_dff.py, actually if  I copy the command that fails when running with pytest and paste in the terminal to execute, it works like a charm....I really don't understand why the same xrun command when launched by pytest/cocotb-test fails to run

aignacio, 2021-02-25:
If I add to tox.ini setenv this:
HOME = /home/user

I see another error:
```bash
-.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib cocotbvhpi_ius: cocotbvhpi_ius: cannot open shared object file: No such file or directory
INFO     cocotb:simulator.py:255 cocotb: Error loading shared library cocotbvhpi_ius
```
I'm start thinking that's related to the way the pytest call the test...

cmarqu, 2021-02-26:
What might give a clue is exporting/extending the environment variable `XRUNOPTS` with the value `"-createdebugscript ~/dbg.csh"` which will write two files, `~/dbg.csh` and `~/dbg.csh.env` where the exact environment and call options of xrun are recorded.

aignacio, 2021-02-26:
Thanks @cmarqu,
apparently the command that I have available here is -debugscript, anyway I ran it with that and here are the env file...not sure what to look for. Do you have a guess?
[dbg.zip](https://github.com/themperek/cocotb-test/files/6049773/dbg.zip)

themperek, 2021-02-28:
> If I add to tox.ini setenv this:
> HOME = /home/user
> 
> I see another error:
> 
> ```shell
> -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib cocotbvhpi_ius: cocotbvhpi_ius: cannot open shared object file: No such file or directory
> INFO     cocotb:simulator.py:255 cocotb: Error loading shared library cocotbvhpi_ius
> ```
> 
> I'm start thinking that's related to the way the pytest call the test...

This is an error if you are using cocotb 1.5 (there ware some changes). In this case, you need to use this https://github.com/themperek/cocotb-test/pull/128 I will realise when cocotb release.

aignacio, 2021-03-01:
Hi @themperek 
I'm trying to install these two version but I'm seeing a message of incompatible issue with pip during tox runtime...
  git+https://github.com/themperek/cocotb-test.git@origin/cocotb_1p5#egg:cocotb-test
  git+https://github.com/cocotb/cocotb.git@stable/1.5#egg:cocotb

```python
ERROR: Cannot install cocotb 1.5.0rc2.dev1+gbd98f994 (from git+https://github.com/cocotb/cocotb.git@stable/1.5#egg:cocotb) and cocotb-test==0.1.2.dev0 because these package versions have conflicting dependencies.
The conflict is caused by:
    The user requested cocotb 1.5.0rc2.dev1+gbd98f994 (from git+https://github.com/cocotb/cocotb.git@stable/1.5#egg:cocotb)
    cocotb-test 0.1.2.dev0 depends on cocotb>=1.5
```

themperek, 2021-03-02:
> I'm trying to install these two version but I'm seeing a message of incompatible issue with pip during tox runtime...

Maybe this: https://stackoverflow.com/questions/12759761/python-pip-force-install-ignoring-dependencies

Seems `>=1.5` does not consider `1.5.0rc2` valid?

aignacio, 2021-03-02:
Apparently it does not...I forked cocotb-test and did some changes to make it work, although I cannot run bc I see now issues with licenses once multiple instances are dispatched, even with tox -- -n 1.