themperek/cocotb-test #130: AttributeError: 'ElementTree' object has no attribute 'getiterator' when using --cocotbxml=... with Python 3.9.1

amal-khailtash, 2021-03-05:
With Python 3.9.1 I get the following error without

```
Traceback (most recent call last):
  ...
  File ".../cocotb_test/plugin.py", line 40, in pytest_sessionfinish
    for testsuite in tree.getiterator("testsuite"):
AttributeError: 'ElementTree' object has no attribute 'getiterator'
```

Based on the following bugfixes:

- [What’s New In Python 3.9](https://docs.python.org/3/whatsnew/3.9.html?highlight=getiterator)
Methods getchildren() and getiterator() of classes ElementTree and Element in the ElementTree module have been removed. They were deprecated in Python 3.2. Use iter(x) or list(x) instead of x.getchildren() and x.iter() or list(x.iter()) instead of x.getiterator(). (Contributed by Serhiy Storchaka in [bpo-36543](https://bugs.python.org/issue36543).)
- [bpo-36543](https://bugs.python.org/issue36543) Removed methods Element.getchildren(), Element.getiterator() and ElementTree.getiterator() and the xml.etree.cElementTree module.

Element.getiterator() should be replaced with .iter(x).  Changing these two lines in `cocotb_test/plugin.py`:

```
40:                for testsuite in tree.getiterator("testsuite"):
...
44:                     for testcase in testsuite.getiterator("testcase"):
```

to:

```
40:                for testsuite in tree.iter("testsuite"):
...
44:                     for testcase in testsuite.iter("testcase"):
```

fixes the issue.

themperek, 2021-03-05:
This is already there: https://github.com/themperek/cocotb-test/blob/9bd8e5a25f4b59b717e80702aeb619b81701def8/cocotb_test/plugin.py#L40-L43

I can make the last release for cocotb 1.4

themperek, 2021-03-05:
I am closing this but feel free to reopen if needed.

closed 2021-03-05 by themperek

amal-khailtash, 2021-03-05:
Yes, you are right, I think I encountered this with coco-tb 1.4.  Thanks.

amal-khailtash, 2021-03-05:
I get this message though when I try to install cocotb-test.  Is cocotb-test compatible with cocotb>=1.4?  You seem to have a hard requirement on cocotb==1.4.

`cocotb-test 0.1.2.dev0 requires cocotb==1.4.*, but you have cocotb 1.5.0rc2 which is incompatible.`

And cocotb-bus 0.1.1 seems to requires cocotb<2.0,>=1.5.0.dev, so it only gets cocotb=1.4.0 and not cocotb=1.5.0rc2.  Is it possible to relax cocotb-test requirement to be cocotb>=1.4?

themperek, 2021-03-05:
Unfortunately, there are some incompatibilities so I did this to avoid issues.
This need to get in https://github.com/themperek/cocotb-test/pull/128 for 1.5 support. 

All this will get cleaned next week when cocotb 1.5 is released. Sorry for the inconvinicen.
I will also release new versions to the cocotb-test next week.

amal-khailtash, 2021-03-05:
That is alright, thanks for the info and this package. :smile:

amal-khailtash, 2021-03-12:
I confirm that the latest master branch works with cocotb 1.5 and modified pytest-parallel.