themperek/cocotb-test #136: pip install fails on branch cocotb_1p5

sjalloq, 2021-03-11:
I'm trying to `pip install` the 1.5 branch and it's failing on the dependency check.  I've got 1.5 installed as shown below.  Any poniters as to what is wrong?

```
ERROR: Could not find a version that satisfies the requirement cocotb>=1.5 (from cocotb-test)
ERROR: No matching distribution found for cocotb>=1.5
(venv) [ShareefJalloq@asic cocotb-test]$ pip list | grep cocotb
cocotb             1.5.0.dev305+g427a0af
```

sjalloq, 2021-03-11:
Same with 1.5.0rc3

themperek, 2021-03-11:
We need a release that should happen today. Or ignore those dependencies.

sjalloq, 2021-03-11:
How can I force the install?  Is this just a pip issue in how it checks a version flag somewhere?

sjalloq, 2021-03-11:
Sorry, should have RTFM: `--no-deps`.

closed 2021-03-11 by sjalloq