themperek/cocotb-test #137: simulation.run can't handle whitespace in arguments

sjalloq, 2021-03-11:
Not sure if this is by design or not, but it took me a while to track down why `vsim` was not accepting my `sim_args`.  It looks like you escape white space but don't remove the escape parameters before invoking `vsim`.

`sim_args=[ "-l logfile.log" ]` results in `vsim -l\ logfile.log`.

To fix this I seem to need to specify each argument individually: `sim_args=[ "-l", "logfile.log" ]`

themperek, 2021-03-12:
Would need `.split()` all commands string that goes here :
https://github.com/themperek/cocotb-test/blob/f31e43d5c15392dc976a01eccb7bc5ed6a72c790/cocotb_test/simulator.py#L252-L253

sjalloq, 2021-03-12:
Sorry, not sure what you're saying.  Are you suggesting a workaround or stating a reason why it's done that way?

themperek, 2021-03-12:
One way to solve this problem.

alexforencich, 2023-01-18:
I think this is just how arguments are passed to processes.  If you want two arguments, you use two items in the list.  This removes the ambiguity of spaces in file names and what not - if it's one entry in the list, it's one argument, even if it contains spaces.  So, IMO this is cocotb-test working exactly as it should.

themperek, 2023-01-19:
@alexforencich is right. It is a feutre not a bug. I will close this in case please feel free to reopen.

closed 2023-01-19 by themperek