themperek/cocotb-test #138: [Enhancement] Specify VCS output executable instead of simv for parallel runs

amal-khailtash, 2021-03-11:
Using pytest-xdist or pytest-parallel would not work with single generated **simv**.  Specifying a different **work_dir** does not seem to work either.

Having a way to separate VCS runs (output to a different directory) or a different name for **simv** would solve this issue.

themperek, 2021-03-12:
Something like this?
https://github.com/themperek/cocotb-test/blob/b992e3c3c7644f8581811cd2ca6a01d23f2115b3/tests/test_parameters.py#L53-L54

amal-khailtash, 2021-03-12:
I guess I was confused between work_dir and sim_build.  That could work, but if I wanted to create all tests in the same directory, this would not let me do that.  Still a good work around.  Thanks.