themperek/cocotb-test #140: Full working sample with toplevel file missing

namoscagnm, 2021-04-05:
Hi,

I have experience in Python, but I am new to Verilog and I am having some problems trying to create a working test. The problem I am having is around the issue of the toplevel. As I didnt find any pre-defined example of a toplevel I am trying to create my own, and having all types of problems with it.

As I am a newbie in Verilog I am probably doing some very basic error, but I cant find out what. Would it be possible for you to put a sample with a complete toplevel file in your instructions and the corresponding Verilog implementation of the modules, so I can have a better guidance into the proper direction?

As I have icarus, I run the simulation with the following command:
`SIM=icarus  pytest -o log_cli=True test_dff.py     `                                                                                       

I have a simple D-Flop module implemented in Verilog and you can see the errors Im getting below,:

[image]
[image]
[image]
[image]

I did my flipflop module as in this file:

dff.v
```
module dff(clk, d, q);
    input clk;
    input d;
    output q;
    
    reg q=0;

   

    always @(posedge clk)
    begin
        q <= d;
    end

endmodule
```

dff_cocotb.py:
```
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import FallingEdge

@cocotb.test()
async def test_dff_simple(dut):
    """ Test that d propagates to q """

    clock = Clock(dut.clk, 10, units="us")  # Create a 10us period clock on port clk
    cocotb.fork(clock.start())  # Start the clock

    await FallingEdge(dut.clk)  # Synchronize with the clock
    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val  # Assign the random value val to the input port d
        await FallingEdge(dut.clk)
        assert dut.q.value == val, "output q was incorrect on the {}th cycle".format(i)
```

test_dff.py:
```
from cocotb_test.simulator import run
import pytest
import os

tests_dir = os.path.dirname(__file__)

@pytest.mark.skipif(os.getenv("SIM") == "ghdl", reason="Verilog not suported")
# @pytest.mark.parametrize("seed", range(10))
def test_dff_verilog():
    run(
        verilog_sources=[
            os.path.join(tests_dir, "dff.v"),
            # os.path.join(tests_dir, "top.v"),
        ],
        toplevel="dff",
        module="dff_cocotb",
    )  # sources  # top level HDL  # name of cocotb test module
```

Thank you so much!

themperek, 2021-04-05:
I think I would need a full log file.

themperek, 2022-02-04:
I close for now. Please re-open if need help.

closed 2022-02-04 by themperek