themperek/cocotb-test #147: "AssertionError: Simulation terminated abnormally. Results file not found" with verilator

kevinyuan, 2021-06-29:
Dear cocotb-test team,

I see the following error when I run cocotb-test with verilator , but everything OK with icarus.

cmdline:
SIM=verilator pytest -o log_cli=True --junitxml=test-results.xml --cocotbxml=test-cocotb.xml

log:

../../../.tox/py36/lib/python3.6/site-packages/cocotb_test/simulator.py:941: in run
    return sim.run()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <cocotb_test.simulator.Verilator object at 0x7f63593802e8>

    def run(self):

        sys.tracebacklimit = 0  # remove not needed traceback from assert

        # use temporary results file
        if not os.getenv("COCOTB_RESULTS_FILE"):
            tmp_results_file = tempfile.NamedTemporaryFile(prefix=self.sim_dir + os.path.sep, suffix="_results.xml")
            results_xml_file = tmp_results_file.name
            tmp_results_file.close()
            self.env["COCOTB_RESULTS_FILE"] = results_xml_file
        else:
            results_xml_file = os.getenv("COCOTB_RESULTS_FILE")

        cmds = self.build_command()
        self.set_env()
        self.execute(cmds)

        if not self.compile_only:
            results_file_exist = os.path.isfile(results_xml_file)
>           assert results_file_exist, "Simulation terminated abnormally. Results file not found."
E           AssertionError: Simulation terminated abnormally. Results file not found.
E           assert False

../../../.tox/py36/lib/python3.6/site-packages/cocotb_test/simulator.py:221: AssertionError

themperek, 2021-07-01:
You have to look more. This message just means that there is no result file. There was something wrong with starting the simulation? 
One would need a full log since there should be more information.

wqueree, 2021-12-20:
I get the same issue using VHDL and cocotb-test 0.2.1 on Fedora 30. It seems to be an issue where `tempfile` is unable to create the file in the `sim_build` directory persistently. Modification of `simulator.Simulator.run` to create a file of the same name with `open` results in this issue not arising.

themperek, 2021-12-20:
Could any of you remove ` sys.tracebacklimit = 0 ` and check what is the error message?

wqueree, 2021-12-20:
The same as above from @kevinyuan: `AssertionError: Simulation terminated abnormally. Results file not found.`  

As far as I can tell, there is no additional traceback logging to the console when `sys.tracebacklimit = 0` is removed.

themperek, 2022-02-04:
@kevinyuan  Can you check with the current master?

themperek, 2022-02-18:
I close this if it still needs attention please reopen.

kschoos, 2022-06-30:
Hi, I do have this problem too.
I debugged it and calling `tempfile.NamedTemporaryFile(prefix=self.sim_dir + os.path.sep, suffix="_results.xml")` and `tmp_results_file.close()` will create the file first and then delete it when closing it. Thus `results_file_exist = os.path.isfile(results_xml_file)` always fails in that scenario.

If I add `delete=False`, it works:

`tempfile.NamedTemporaryFile(prefix=self.sim_dir + os.path.sep, suffix="_results.xml", delete=False)`

themperek, 2022-07-01:
@kschoos Can you make PR for this?

closed 2022-07-01 by themperek