themperek/cocotb-test #151: ResourceWarning: unclosed file

pytricke, 2021-09-14:
At the end of every simulation (like the test_dff.py example) , I get the following warning:

```
INFO     cocotb:simulator.py:267 sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name=2 mode='w' encoding='UTF-8'>
```

The name=2 part suggests an unclosed reference to stderr.
It seems to be caused by the Simulator.execute function:
```

            with subprocess.Popen(
                cmd,
                stdout=subprocess.PIPE,
                stderr=subprocess.STDOUT,
                cwd=self.work_dir,
                env=self.env
            ) as p:
                self.process = p
                for line in p.stdout:
                    self.logger.info(line.decode("utf-8").rstrip())
```
More specifically, if I comment out the line:
```
                stderr=subprocess.STDOUT,
```
the ResourceWarning is no longer issued.

As far as I can tell, this warning only occurs when using python verions >= 3.9.
Any thoughts on this?

Please let me know if more information is needed.

Thanks,
Patrick

themperek, 2021-09-16:
>Any thoughts on this?
No idea I will try to have a look.

themperek, 2021-09-17:
Maybe call `self.process.terminate()` somewhere https://github.com/themperek/cocotb-test/blob/7ade4fe5c4f8665095963715f2fcc8b8b94d14dd/cocotb_test/simulator.py#L269

pytricke, 2021-09-20:
Thanks a lot for the feedback!
When I have some more time, I'll definitely do some more digging and report back if I find anything worth mentioning.

pytricke, 2021-09-20:
I tried a few things and it seems that the warning is only printed (i.e. brought to light) by cocotb_test, but seems to be caused by cocotb.
Using version 1.5.0 of cocotb I no longer get the ResourceWarning (which I do get when using cocotb versions 1.5.1 or 1.5.2).

I am closing this issue as it apparently is not related to cocotb_test.

Thanks again for your feedback!

closed 2021-09-20 by pytricke