themperek/cocotb-test #155: Questa vcom does not know "-incr"

benbr8, 2022-01-03:
It seems to be verilog (`vlog`) only, I checked the documentation.

```
INFO     cocotb:simulator.py:281 #  vcom -mixedsvvh -incr -work some_lib  /path/to/cocotb-test/tests/dff.vhdl
INFO     cocotb:simulator.py:281 # QuestaSim-64 vcom 2021.1_1 Compiler 2021.02 Feb 21 2021
INFO     cocotb:simulator.py:281 # ** Error (suppressible): (vcom-1902) Option "-incr" is not a valid option for this tool.
```

themperek, 2022-01-03:
Yes, this is a bug.

>Was there any reason not using Simulator.outdated() for Questa, other than having a built-in function available?

In principle, it is better and it will check all dependencies also automatic included but it has many issues and I think this should be rewarded.

benbr8, 2022-01-24:
Issue was fixed in #166, however now Questa VHDL sources will be compiled every time, outdated or not.

closed 2022-01-24 by benbr8