themperek/cocotb-test #160: Persistent compile output in `sim_build` causes tests to pass, even if compilation failed

benbr8, 2022-01-05:
Persisting successfully compiled libraries in `sim_build` retain their validity, even after a changed source file does not compile successfully.

This issue exists for Questa, and I could imagine that Aldec simulators are affected as well. Currently I do not have a solution other than deleting compiled libs in `sim_build` prior to compilation attempt.

themperek, 2022-01-05:
This is an issue I have no solution to. Not sure what can be done here. `cocotb-clean` is something I used if I change the Verilog code.

jonpovey, 2022-01-14:
This may be fixed by my PR #164 which fixes silently ignored compile failures (at least for Modelsim - should fix Questa too)

themperek, 2022-01-14:
OK. I was thinking about something else. This is a bug that can be fixed.

themperek, 2022-02-04:
@benbr8 This should be fixed?

benbr8, 2022-02-10:
Yes, test execution is now failing as it should. Thanks :)