themperek/cocotb-test #161: How to keep questa session running when simulation is done

filamoon, 2022-01-12:
This is a follow up for https://github.com/cocotb/cocotb/issues/2829

I tried both GUI=1 when calling "make" and "gui=True" when calling cocotb_test.simulator.run, but the questasim gui still quits when simulation is done...

====
make SHELL='sh -xv' WAVES=1 GUI=1

====
    cocotb_test.simulator.run(
        python_search=[tests_dir],
        verilog_sources=verilog_sources,
        toplevel=toplevel,
        module=module,
        parameters=parameters,
        sim_build=sim_build,
        extra_env=extra_env,
        gui=True,
    )

===
At the end of log, there is a call to "$finish". I also added "-onfinish ask" when invoking vsim but still it didn't work.

# ** Note: $finish

themperek, 2022-01-14:
If `GUI` is set it should not finish:

https://github.com/themperek/cocotb-test/blob/19b1cd087407926005ac8c865934455e913b8b76/cocotb_test/simulator.py#L475-L476

jonpovey, 2022-01-25:
I am using Modelsim (works the same as Questa) with the `gui` parameter and the sim stays open for me a the end of tests. I am using it like so with the latest cocotb-test master:
```
gui = os.getenv("GUI", False)
```

themperek, 2022-02-18:
I close this if it still needs attention please reopen.

closed 2022-02-18 by themperek