themperek/cocotb-test #169: Questa HDL code coverage export

benbr8, 2022-01-24:
This one I'm not so sure on how to handle, so I'd like to have your input.

With Questa, getting the simulator to export code coverage data requires multiple things:

1. pass `-coveropt 1` to compiler -> can be done via `compile_args`
2. pass `-coverage` to sim command -> can be done via `sim_args`
3. execute command `coverage save -onexit some_name.ucdb;` in TCL shell after calling `vsim` like with `waves=True`

Currently there is no way to do (3). Solutions could be to either
* create a `coverage` parameter for `Simulator` which would handle all of the above or only (3). This could be used to implement similar functionality for other simulators.

or
* add possibility to add custom do-script extensions (applies only to Questa and Aldec Simulators)

Let me know what you think

themperek, 2022-01-24:
To not complicate things I would think the way to go is extend `Questa` class for such special cases.

See: https://github.com/themperek/cocotb-test/blob/841486fb83ab1c590dddaa8304312acc7830f072/tests/test_dff_custom_sim.py#L9-L21

Maybe before one should split `build_command()` in smaller functions so it is handy. See `Activehdl` class.

More: https://github.com/themperek/cocotb-test/pull/152

Would this work?

benbr8, 2022-01-24:
Sure, let me see if I can come up with something useful

benbr8, 2022-01-25:
If #171 gets approved, my use case can be implemented like this:
```python
class QuestaCoverage(Questa):
    def __init__(self, coverage=False, *argv, **kwargs):
        super().__init__(*argv, **kwargs)

        self.coverage = coverage

    def build_script_sim(self):
        sim_script = super().build_script_sim()
        sim_script += f"coverage save -onexit {self.module}.ucdb;"
        return sim_script

def test_dff_coverage():
    QuestaCoverage(
        verilog_sources=[os.path.join(hdl_dir, "dff.v")],
        toplevel="dff_test",
        python_search=[hdl_dir],
        module="dff_cocotb",
        compile_args=["-coveropt", "1"],
        sim_args=["-coverage"],
        coverage=True,
    ).run()
```

benbr8, 2022-02-04:
Implemented with #171, thanks for your support :)

closed 2022-02-04 by benbr8