themperek/cocotb-test #179: GHDL and cocotb-test

adi8v, 2022-03-01:
I'm trying to use cocotb-test with GHDL to allow testing multiple times with varying VHDL generics. If I run a single test of the dut using cocotb all is (reasonably) well:

`(base) C:\git\dev_aph\firmware\common\utils\led_flasher\tests>make
make -f Makefile results.xml
make[1]: Entering directory '/c/git/dev_aph/firmware/common/utils/led_flasher/tests'
\
/c/eda/ghdl/GHDL/MINGW32-mcode-standalone/bin/ghdl -i    --workdir=sim_build --work=work /c/git/dev_aph/firmware/common/utils/led_flasher/hdl/led_flasher.vhd && \
/c/eda/ghdl/GHDL/MINGW32-mcode-standalone/bin/ghdl -m    --workdir=sim_build -Psim_build --work=work led_flasher
MODULE=led_flasher_test TESTCASE= TOPLEVEL=led_flasher TOPLEVEL_LANG=vhdl \
 /c/eda/ghdl/GHDL/MINGW32-mcode-standalone/bin/ghdl -r   --workdir=sim_build -Psim_build --work=work led_flasher --vpi=C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll --ieee-asserts=disable --wave=output.ghw
**loading VPI module 'C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'**
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     Running on GHDL version 2.0.0-dev (1.0.0.r1012.g8fb84c1c) [Dunoon edition]
     0.00ns INFO     Running tests with cocotb v1.6.2 from C:\Users\aphol\miniconda3_32\lib\site-packages\cocotb
     0.00ns INFO     Seeding Python random module with 1646147419
     0.00ns INFO     Found test led_flasher_test.run_test_001
     0.00ns INFO     ←[34mrunning←[49m←[39m run_test_001 (1/1)
vpi_get: unknown property
     0.00ns INFO     count_index = 4
     0.00ns INFO     resetting the module
    50.00ns INFO     out of reset
  1050.00ns INFO     done
  1050.00ns INFO     run_test_001 ←[32mpassed←[49m←[39m
  1050.00ns INFO     ***************************************************************************************
                     ** TEST                           STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                     ***************************************************************************************
                     ** led_flasher_test.run_test_001  ←[32m PASS ←[49m←[39m       1050.00           0.01     102454.79  **
                     ***************************************************************************************
                     ** TESTS=1 PASS=1 FAIL=0 SKIP=0                1050.00           0.05      20758.60  **
                     ***************************************************************************************

make[1]: Leaving directory '/c/git/dev_aph/firmware/common/utils/led_flasher/tests'

(base) C:\git\dev_aph\firmware\common\utils\led_flasher\tests>`

However, if I try to expand that using a parameterised pytest:

`(base) C:\git\dev_aph\firmware\common\utils\led_flasher\tests>pytest test_led_flasher.py
=========================================================================================================== test session starts ============================================================================================================
platform win32 -- Python 3.9.7, pytest-7.0.1, pluggy-1.0.0
rootdir: C:\git\dev_aph\firmware\common\utils\led_flasher\tests
plugins: cocotb-test-0.2.2
collected 2 items

test_led_flasher.py FF                                                                                                                                                                                                                [100%]

================================================================================================================= FAILURES =================================================================================================================
___________________________________________________________________________________________________________ test_led_flasher[3] ____________________________________________________________________________________________________________

request = <FixtureRequest for <Function test_led_flasher[3]>>, count_index = 3

    @pytest.mark.parametrize("count_index", [3 ,4])
    def test_led_flasher(request, count_index):
        dut = "led_flasher"
        module = "test_led_flasher" #os.path.splitext(os.path.basename(__file__))[0]
        toplevel = dut
        print("Module is:")
        print(module)

        vhdl_sources = [
            os.path.join(tests_dir, '..' , 'hdl', f"{dut}.vhd"),
        ]

        parameters = {}

        parameters['count_index'] = count_index

        extra_env = {f'PARAM_{k}': str(v) for k, v in parameters.items()}
[... 58 lines cut ...]
            sim_build=sim_build,
            extra_env=extra_env,
        )
E       SystemExit: Process 'ghdl' termindated with error 1

test_led_flasher.py:85: SystemExit
----------------------------------------------------------------------------------------------------------- Captured stdout call -----------------------------------------------------------------------------------------------------------
Module is:
test_led_flasher
------------------------------------------------------------------------------------------------------------ Captured log call -------------------------------------------------------------------------------------------------------------
INFO     cocotb:simulator.py:291 Running command: ghdl -i --work=led_flasher C:\git\dev_aph\firmware\common\utils\led_flasher\hdl\led_flasher.vhd
INFO     cocotb:simulator.py:291 Running command: ghdl -m --work=led_flasher led_flasher
INFO     cocotb:simulator.py:291 Running command: ghdl -r --work=led_flasher led_flasher --vpi=C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll -gcount_index=4
INFO     cocotb:simulator.py:302 loading VPI module 'C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
INFO     cocotb:simulator.py:302 The specified module could not be found.
INFO     cocotb:simulator.py:302
INFO     cocotb:simulator.py:302 C:\eda\ghdl\GHDL\MINGW32-mcode-standalone\bin\ghdl.exe:error: cannot load VPI module 'C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll '
========================================================================================================= short test summary info ==========================================================================================================
FAILED test_led_flasher.py::test_led_flasher[3] - SystemExit: Process 'ghdl' termindated with error 1
FAILED test_led_flasher.py::test_led_flasher[4] - SystemExit: Process 'ghdl' termindated with error 1
============================================================================================================ 2 failed in 0.36s =============================================================================================================

(base) C:\git\dev_aph\firmware\common\utils\led_flasher\tests>`

Looks like it's not finding the GHDL VPI even though it points to the same .dll in both cases.

themperek, 2022-03-02:
I will try to check but hard to say. Why you are using the 32bit version?

adi8v, 2022-03-03:
I'd not managed to get the 64bit (llvm) version of GHDL working so had used the 32bit (mcode) version as I was trying to use v0.37 of GHDL to match another machine. I'll try having a look at the 64bit version again.

themperek, 2022-03-04:
It looks like you need to add: `$(GHDL_IR)/lib` to `PATH`. I will try to patch this (but something is strange for now).

closed 2022-03-04 by themperek

adi8v, 2022-03-07:
Thanks. Got a different error now:

```
-------------------------------------------------------------------------------- Captured stdout call --------------------------------------------------------------------------------
Module is:
test_led_flasher
--------------------------------------------------------------------------------- Captured log call ----------------------------------------------------------------------------------
INFO     cocotb:simulator.py:315 Running command: ghdl -i --work=led_flasher C:\git\dev_aph\firmware\common\utils\led_flasher\hdl\led_flasher.vhd
INFO     cocotb:simulator.py:315 Running command: ghdl -m --work=led_flasher led_flasher
INFO     cocotb:simulator.py:315 Running command: ghdl -r --work=led_flasher led_flasher --vpi=C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll -gcount_index=3
ERROR    cocotb:simulator.py:291 loading VPI module 'C:/Users/aphol/miniconda3_32/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
INFO     cocotb:simulator.py:291      -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
INFO     cocotb:simulator.py:291      -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
ERROR    cocotb:simulator.py:291 VPI module loaded!
INFO     cocotb:simulator.py:291      0.00ns INFO     Running on GHDL version 2.0.0-dev (1.0.0.r1012.g8fb84c1c) [Dunoon edition]
INFO     cocotb:simulator.py:291      0.00ns INFO     Running tests with cocotb v1.6.2 from C:\Users\aphol\miniconda3_32\lib\site-packages\cocotb
INFO     cocotb:simulator.py:291      0.00ns INFO     Seeding Python random module with 1646654821
INFO     cocotb:simulator.py:291      0.00ns CRITICAL Failed to import module test_led_flasher: No module named 'test_led_flasher'
INFO     cocotb:simulator.py:291      0.00ns INFO     MODULE variable was "test_led_flasher"
INFO     cocotb:simulator.py:291      0.00ns INFO     Traceback:
INFO     cocotb:simulator.py:291      0.00ns INFO     Traceback (most recent call last):
INFO     cocotb:simulator.py:291                        File "C:\Users\aphol\miniconda3_32\lib\site-packages\cocotb\regression.py", line 172, in _discover_tests
INFO     cocotb:simulator.py:291                          module = _my_import(module_name)
INFO     cocotb:simulator.py:291                        File "C:\Users\aphol\miniconda3_32\lib\site-packages\cocotb\regression.py", line 66, in _my_import
INFO     cocotb:simulator.py:291                          mod = __import__(name)
INFO     cocotb:simulator.py:291                      ModuleNotFoundError: No module named 'test_led_flasher'

```

However I can "fix" this by running:
`python -m pytest`
rather than just 'pytest'

themperek, 2022-03-07:
I am guessing you need to add the`test_led_flasher.py` path to `python_search` argument.

PR #180 (2022-03-04): Fix GHDL on Windows #179
Closes #179