themperek/cocotb-test #181: Verilator string parameters not generated correctly

rishubn, 2022-03-22:
Verilator requires string parameters to be surrounded by quotes: 
`-Gparam="value"`

> Strings
> 
>     Strings must be in double quotes (“”). They must be escaped properly on the command line, e.g. as -GSTR="\"My String\"" or -GSTR='"My String"'

If given a dict like `{ "param" : "value" }` cocotb_test will generate `-Gparam=value` and Verilator will fail to compile. 

Some simple logic can be added to do this wrapping in simulator.py:
```python
for name, value in parameters.items():
            parameters_cmd.append("-G" + name + "=\"" + str(value) + "\"") if type(value) is str else parameters_cmd.append("-G" + name + "=" + str(value))
```

themperek, 2022-08-19:
@rishubn Does #200 work for you?

themperek, 2022-08-19:
Adding quotes fails on my local machine. Maybe I am doing something wrong.