themperek/cocotb-test #184: Verilator configuration file not recognized

aignacio, 2022-06-06:
Hi @themperek,

do you know how can I add an input configuration file for **verilator**? According to the [verilator manual](https://www.veripool.org/ftp/verilator_doc.pdf), it's possible with `-f verilator.flags` but when I add this option in `compile_args` it shows the following error....For instance [here](https://github.com/aignacio/mpsoc_example/blob/master/makefile#L93) there is an example of how it works while running **verilator** standalone.

```bash
INFO     cocotb:simulator.py:291 Running command: perl /usr/local/bin/verilator -cc --exe -Mdir /axi_dma/run_dir/run_verilator_test_dma_basic_32 -DCOCOTB_SIM=1 --top-module tb_axi_dma --vpi --public-flat-rw --prefix Vtop -o tb_axi_dma -LDFLAGS -Wl,-rpath,/axi_dma/.tox/python/lib/python3.8/site-packages/cocotb/libs -L/axi_dma/.t
ox/python/lib/python3.8/site-packages/cocotb/libs -lcocotbvpi_verilator -f verilator.flags --trace-fst --coverage --trace-structs --Wno-UNOPTFLAT --Wno-REDEFMACRO -DAXI_ADDR_WIDTH=32 -DAXI_DATA_WIDTH=32 -I/axi_dma/rtl/inc -I/axi_dma/rggen-verilog-rtl /axi_dma/.tox/python/lib/python3.8/site-packages/cocotb/share/lib/verilator/verilator.cpp /axi_dma/rtl/inc/axi_pkg.svh /axi_dma/rtl/inc/utils_pkg.svh /axi_dma/rtl/axi_dma_wrapper.sv /axi_dma/rtl/tb_axi_dma.sv /axi_dma/csr_out/csr_dma.v /axi_dma/rggen-verilog-rtl/rggen_adapter_common.v /axi_dma/rggen-verilog-rtl/rggen_address_decoder.v /axi_dma/rggen-verilog-rtl/rggen_apb_adapter.v /axi_dma/rggen-verilog-rtl/rggen_apb_bridge.v /axi_dma/rggen-verilog-rtl/rggen_axi4lite_adapter.v /axi_dma/rggen-verilog-rtl/rggen_axi4lite_bridge.v /axi_dma/rggen-verilog-rtl/rggen_axi4lite_skid_buffer.v /axi_dma/rggen-verilog-rtl/rggen_bit_field.v /axi_dma/rggen-verilog-rtl/rggen_bit_field_w01trg.v /axi_dma/rggen-verilog-rtl/rggen_default_register.v /axi_dma/rggen-verilog-rtl/rggen_external_register.v /axi_dma/rggen-verilog-rtl/rggen_indirect_register.v /axi_dma/rggen-verilog-rtl/rggen_mux.v /axi_dma/rggen-verilog-rtl/rggen_register_common.v /axi_dma/rggen-verilog-rtl/rggen_wishbone_adapter.v /axi_dma/rggen-verilog-rtl/rggen_wishbone_bridge.v
INFO     cocotb:simulator.py:302 %Error: Invalid option: -f verilator.flags
```

themperek, 2022-06-07:
I expect you need to make a list:
```python3
compile_args = ["-f", "verilator.flags"]
```
Maybe should make a change so another way would work too 🤔

aignacio, 2022-06-07:
It does work when adding as you suggest + abs path of the flags file. Thanks =)

**verilator.flags**
```bash
-Wno-UNOPTFLAT
-Wno-TIMESCALEMOD
....
/axi_dma [Full path of container image]/verilator_config.vlt
```

**verilator_config.vlt**
```bash
`verilator_config
lint_off -rule WIDTH    -file "**/rggen_register_common.v"
lint_off -rule WIDTH    -file "**/rggen_adapter_common.v"
lint_off -rule UNSIGNED -file "**/rggen_address_decoder.v"
lint_off -rule WIDTH    -file "**/csr_dma.v"
```

closed 2022-06-07 by aignacio