themperek/cocotb-test #185: Running cocotb-test

TomO700, 2022-06-08:
I am trying to modify the default python and run with the following command using verilator, and end up with the error below

_Command_

`SIM=verilator pytest -o log_cli=True test_mod.py`

_Error_

```
E       SystemExit: Process 'perl' terminated with error 2

/../../.local/lib/python3.9/site-packages/cocotb/runner.py:134: SystemExit
```

Any help would be highly appreciated

themperek, 2022-06-09:
Full log file please with:
`SIM=verilator pytest -s -o log_cli=True test_mod.py`

TomO700, 2022-06-09:
Hi,

Please find the log below.

Thanks a lot

============================= test session starts
==============================
platform linux -- Python 3.9.5, pytest-7.1.2, pluggy-1.0.0
rootdir: /mnt/D/d/OB_tb
plugins: cocotb-test-0.2.2
collected 1 item

test_ob.py::test_simple_ob_runner INFO: Running command: perl
"/usr/local/bin/verilator "-cc "--exe "-Mdir "/mnt/D/d/OB_tb/sim_build
"-DCOCOTB_SIM=1 "--top-module "ob_top "--vpi "--public-flat-rw "--prefix
"Vtop "-o "ob_top "-LDFLAGS
"-Wl,-rpath,/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs
-L/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs
-lcocotbvpi_verilator
"/home/tomo/.local/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp
"/mnt/D/d/OB_tb/ob_pkg.sv "/mnt/D/d/OB_tb/ob_top.sv "/mnt/D/d/OB_tb/
update.sv "/mnt/D/d/OB_tb/lookup.sv "/mnt/D/d/OB_tb/add.sv "/mnt/D/d/OB_tb/
cell.sv "/mnt/D/d/OB_tb/cell_sort.sv "/mnt/D/d/OB_tb/mod.sv "/mnt/D/d/OB_tb/
del.sv" in directory:"/mnt/D/d/OB_tb/sim_build"
FAILED

=================================== FAILURES
===================================
____________________________ test_simple_ob_runner
_____________________________

    def test_simple_ob_runner():

        toplevel_lang = os.getenv("TOPLEVEL_LANG", "verilog")
        sim = os.getenv("SIM", "verilator")

        proj_path = Path(__file__).resolve().parent

        verilog_sources = []
        vhdl_sources = []

        if toplevel_lang == "verilog":
            verilog_sources = [proj_path / "ob_pkg.sv",
                                proj_path / "ob_top.sv",
                                proj_path / "update.sv",
                                proj_path / "lookup.sv",
                                proj_path / "add.sv",
                                proj_path / "cell.sv",
                                proj_path / "cell_sort.sv",
                                proj_path / "mod.sv",
                                proj_path / "del.sv"]
        #else:
            #vhdl_sources = [proj_path / "*.vhdl"]
        runner = get_runner(sim)()
>       runner.build(
            verilog_sources=verilog_sources, vhdl_sources=vhdl_sources,
toplevel="ob_top"
        )

test_ob.py:129:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _

self = <cocotb.runner.Verilator object at 0x7f3568f45670>, library_name =
'work'
verilog_sources = [PosixPath('/mnt/D/d/OB_tb/ob_pkg.sv'),
PosixPath('/mnt/D/d/OB_tb/ob_top.sv'), PosixPath('/mnt/D/d/OB_tb/update.sv'),
PosixPath('/mnt/D/d/OB_tb/lookup.sv'), PosixPath('/mnt/D/d/OB_tb/add.sv'),
PosixPath('/mnt/D/d/OB_tb/cell.sv'), ...]
vhdl_sources = [], includes = [], defines = [], parameters = {}, extra_args
= []
toplevel = 'ob_top', always = False, build_dir = 'sim_build'

    def build(
        self,
        library_name: str = "work",
        verilog_sources: Sequence[PathLike] = [],
        vhdl_sources: Sequence[PathLike] = [],
        includes: Sequence[PathLike] = [],
        defines: Sequence[str] = [],
        parameters: Mapping[str, object] = {},
        extra_args: Sequence[str] = [],
        toplevel: Optional[str] = None,
        always: bool = False,
        build_dir: PathLike = "sim_build",
    ) -> None:
        """Build the HDL sources."""

        self.build_dir = os.path.abspath(build_dir)
        os.makedirs(self.build_dir, exist_ok=True)

        # note: to avoid mutating argument defaults, we ensure that no value
        # is written without a copy. This is much more concise and leads to
        # a better docstring than using `None` as a default in the
parameters
        # list.
        self.library_name = library_name
        self.verilog_sources = get_abs_paths(verilog_sources)
        self.vhdl_sources = get_abs_paths(vhdl_sources)
        self.includes = get_abs_paths(includes)
        self.defines = list(defines)
        self.parameters = dict(parameters)
        self.compile_args = list(extra_args)
        self.always = always
        self.hdl_toplevel = toplevel

        for e in os.environ:
            self.env[e] = os.environ[e]

        cmds = self.build_command()
>       self.execute(cmds, cwd=self.build_dir)
E       SystemExit: Process 'perl' terminated with error 1

/home/tomo/.local/lib/python3.9/site-packages/cocotb/runner.py:134:
SystemExit
=============================== warnings summary
===============================
test_ob.py:12
  /mnt/D/d/OB_tb/test_ob.py:12: UserWarning: Python runners and associated
APIs are an experimental feature and subject to change.
    from cocotb.runner import get_runner

test_ob.py:16
  /mnt/D/d/OB_tb/test_ob.py:16: PytestUnknownMarkWarning: Unknown
pytest.mark.simulator_required - is this a typo?  You can register custom
marks to avoid this warning - for details, see
https://docs.pytest.org/en/stable/how-to/mark.html
    pytestmark = pytest.mark.simulator_required

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=========================== short test summary info
============================
FAILED test_ob.py::test_simple_ob_runner - SystemExit: Process 'perl'
termina...
======================== 1 failed, 2 warnings in 0.16s
=========================

themperek, 2022-06-09:
I am lost what you are using `cocotb-test` or python runner from `cocotb` ?
Maybe it was correct repository for the issue after all. BTW: `cocotb` runner will change a lot still.

It looks like a compilation fail. Not sure why there is no more info in the log file.

In any case please try to run from command line 
```bash
perl /usr/local/bin/verilator -cc --exe -Mdir /mnt/D/d/OB_tb/sim_build -DCOCOTB_SIM=1 --top-module ob_top --vpi --public-flat-rw --prefixsv Vtop -o ob_top -LDFLAGS -Wl,-rpath,/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -L/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -lcocotbvpi_verilator /home/tomo/.local/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp /mnt/D/d/OB_tb/ob_pkg.sv /mnt/D/d/OB_tb/ob_top.sv /mnt/D/d/OB_tb/ update.sv /mnt/D/d/OB_tb/lookup.sv /mnt/D/d/OB_tb/add.sv /mnt/D/d/OB_tb/ cell.sv /mnt/D/d/OB_tb/cell_sort.sv /mnt/D/d/OB_tb/mod.sv /mnt/D/d/OB_tb/ del.sv
```

TomO700, 2022-06-12:
Not sure, if the mails reached you or not. I replies about 3-4 days ago. Will either way copy them below
I tried the command with a few suggested changes to switches, wondering if it's an installation or a path problem.
--prefixsv was changed to -prefix
-L & -l were changed to -D

Below is the command and error 

perl /usr/local/bin/verilator -cc --exe -Mdir /mnt/D/d/OB_tb/sim_build -DCOCOTB_SIM=1 --top-module ob_top --vpi --public-flat-rw --prefix Vtop -o ob_top -LDFLAGS -Wl,-rpath,/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -D /home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -D cocotbvpi_verilator /home/tomo/.local/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp /mnt/D/d/OB_tb/[ob_pkg.sv](http://ob_pkg.sv/) /mnt/D/d/OB_tb/[ob_top.sv](http://ob_top.sv/) /mnt/D/d/OB_tb/ [update.sv](http://update.sv/) /mnt/D/d/OB_tb/[lookup.sv](http://lookup.sv/) /mnt/D/d/OB_tb/[add.sv](http://add.sv/) /mnt/D/d/OB_tb/ [cell.sv](http://cell.sv/) /mnt/D/d/OB_tb/[cell_sort.sv](http://cell_sort.sv/) /mnt/D/d/OB_tb/[mod.sv](http://mod.sv/) /mnt/D/d/OB_tb/ [del.sv](http://del.sv/)
%Error: Cannot find file containing module: /home/tomo/.local/lib/python3.9/site-packages/cocotb/libs
%Error: This may be because there's no search path specified with -I<dir>.
        ... Looked in:
             /home/tomo/.local/lib/python3.9/site-packages/cocotb/libs
             /home/tomo/.local/lib/python3.9/site-packages/cocotb/libs.v
             /home/tomo/.local/lib/python3.9/site-packages/cocotb/[libs.sv](http://libs.sv/)
             /mnt/D/d/OB_tb/sim_build//home/tomo/.local/lib/python3.9/site-packages/cocotb/libs
             /mnt/D/d/OB_tb/sim_build//home/tomo/.local/lib/python3.9/site-packages/cocotb/libs.v
             /mnt/D/d/OB_tb/sim_build//home/tomo/.local/lib/python3.9/site-packages/cocotb/[libs.sv](http://libs.sv/)
%Error: Cannot find file containing module: cocotbvpi_verilator
%Error: Cannot find file containing module: /mnt/D/d/OB_tb/
%Error: Exiting due to 4 error(s)

TomO700, 2022-06-12:
Btw, even for the dff example, the below command fails for verilator, but works with icarus

_SIM=verilator pytest -o log_cli=True test_dff.py_

TomO700, 2022-06-12:
Solved this, reinstalling verilator and also making a change to the $PATH

closed 2022-06-12 by TomO700