themperek/cocotb-test #189: Verilator with cocotb-test - Error

TomO700, 2022-06-16:
Hi,

I get the following error when running verilator, any idea what could be causing this? Previously I have run similar tests without issues

```
Entering directory '/mnt/D/d/count/sim_build'
g++    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_fst_c.o Vtop__ALL.a   -Wl,-rpath,/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -L/home/tomo/.local/lib/python3.9/site-packages/cocotb/libs -lcocotbvpi_verilator -lz    -o countingModule
/usr/bin/ld: verilator.o: in function `main':
verilator.cpp:(.text.startup+0x4b): undefined reference to `Vtop::Vtop(char const*)'
/usr/bin/ld: verilator.cpp:(.text.startup+0xbd): undefined reference to `Vtop::trace(VerilatedFstC*, int, int)'
/usr/bin/ld: verilator.cpp:(.text.startup+0xf5): undefined reference to `Vtop::eval_step()'
/usr/bin/ld: verilator.cpp:(.text.startup+0x189): undefined reference to `Vtop::final()'
collect2: error: ld returned 1 exit status
make: *** [Vtop.mk:67: countingModule] Error 1
make: Leaving directory '/mnt/D/d/count/sim_build'
Process 'make' terminated with error 2
```

Thanks

themperek, 2022-06-16:
Do `cocotb`/"Makfie" examples work?

See: https://docs.cocotb.org/en/stable/simulator_support.html#verilator

TomO700, 2022-06-18:
Makefile examples are fine.
This is a custom design that I am trying. I want to run it with cocotb-test

themperek, 2022-08-19:
I would need an example to say anything.

themperek, 2022-08-19:
I have seen this recently. It does this if there is some other project compiled in the directory above. Do not know what is going on yet.

themperek, 2022-08-19:
Seems a strange situation when compiling with "WAVES" and without. `cocotb-clean -r` and not mixing `WAVES` seems to help. No idea why.