themperek/cocotb-test #193: 'log' command is not supported in the VSimSA DO Mode for Active HDL

jackbpdx, 2022-08-05:
There is a syntax error with the 'log' command so it needs to be changed to the 'trace' command.

themperek, 2022-08-08:
Need more details.

jackbpdx, 2022-08-08:
In the documentation it states that the log command is not supported in DO mode. 
[image]

themperek, 2022-08-09:
It is this?
https://github.com/themperek/cocotb-test/blob/ab9e3eaab10ab033949094ea6503b5be19589396/cocotb_test/simulator.py#L938

Can you change and provide PR? It is a bit difficult with the license for me at the moment.

jackbpdx, 2022-08-09:
That line is referring to Riviera but I'm referring to Active -HDL. 
https://github.com/themperek/cocotb-test/blob/ab9e3eaab10ab033949094ea6503b5be19589396/cocotb_test/simulator.py#L1022

However, I'm sure Riviera has the same issue. 
What is PR?

themperek, 2022-08-10:
>That line is referring to Riviera but I'm referring to Active -HDL.

How it should look like to make it work?

PR -> https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests

closed 2022-08-17 by themperek