themperek/cocotb-test #196: Waveform dump broken for iverilog

stephancill, 2022-08-14:
I am trying to get my verilog file to dump waves when compiled with iverilog. But when I run tests with the `WAVES=1` env variable I get the following error:
```
---------------------------------------------------------------------- Captured log call -----------------------------------------------------------------------
INFO     cocotb:simulator.py:291 Running command: iverilog -o /path/to/repo/cocotb-test-bug-demo/src/sim_build/dff.vvp -D COCOTB_SIM=1 -s dff -g2012 -s iverilog_dump /Users/stephan/environments/personal/pus/cocotb-test-bug-demo/src/dff.v
INFO     cocotb:simulator.py:302 error: Unable to find the root module "iverilog_dump" in the Verilog source.
INFO     cocotb:simulator.py:302      : Perhaps ``-s iverilog_dump'' is incorrect?
INFO     cocotb:simulator.py:302 2 error(s) during elaboration.
```

I think this got broken somewhere down the line after it got implemented because the original implementation in https://github.com/themperek/cocotb-test/pull/116 worked.

Demo repo available here: https://github.com/stephancill/cocotb-test-waves-bug-demo

hansemro, 2022-08-15:
I can confirm latest tagged release (v0.2.2) has this issue with iverilog (regression introduced by 313d51c6737), but seems to have been fixed in later commits. So try installing from repo directly:
```
git clone https://github.com/themperek/cocotb-test.git
pip install -v -e cocotb-test
```

themperek, 2022-08-19:
@hansemro Thank you for the report. Should be fixed.

PR #199 (2022-08-19): Fix waveform dump for icarus #196
Closes #196
closed 2022-08-19 by themperek