themperek/cocotb-test #201: [Enhancement] Allow running of tcl scripts before compilation/simulation steps

alexgomp, 2022-08-24:
**Problem:**
* The current framework enforces a compilation order between VHDL and Verilog sources (e.g. in the case of the Questa simulator, all VHDL sources are compiled before Verilog).  This can be a problem when a VHDL source has a Verilog source as a dependency.
* Many IP blocks provided by Xilinx, Intel, or 3rd parties are provided with a TCL script(s) to aid compilation. Especially in the case of IP with many source files, it would be nice to have a way to easily compile this IP using the provided scripts.

**Proposed Solution**
Add a parameter to the Simulator block, `tcl_sources`:

I've written a solution for Questa/Modelsim:
```
--------------------------- cocotb_test/simulator.py ---------------------------
index f304ae8..c3cc463 100644
@@ -39,6 +39,7 @@ class Simulator:
         work_dir=None,
         python_search=None,
         toplevel_lang="verilog",
+        tcl_sources=None,
         verilog_sources=None,
         vhdl_sources=None,
         includes=None,
@@ -94,6 +95,10 @@ class Simulator:
         self.toplevel = toplevel
         self.toplevel_lang = toplevel_lang
 
+        if tcl_sources is None:
+            tcl_sources = []
+        self.tcl_sources = self.get_abs_paths(tcl_sources)
+
         if verilog_sources is None:
             verilog_sources = []
         self.verilog_sources = self.get_abs_paths(verilog_sources)
@@ -495,6 +500,10 @@ class Questa(Simulator):
 
         cmd = []
 
+        if self.tcl_sources:
+            for src in self.tcl_sources:
+                cmd.append(["vsim", "-do", src])
+
         if self.vhdl_sources:
             compile_args = self.compile_args + self.vhdl_compile_args
 

------------------------------ tests/test_dff.py ------------------------------
index 40a3243..e8a4f8a 100644
@@ -1,3 +1,5 @@
+import pathlib
+
 from cocotb_test.simulator import run
 import pytest
 import os
@@ -17,6 +19,13 @@ def test_dff_vhdl():
     run(vhdl_sources=[os.path.join(tests_dir, "dff.vhdl")], toplevel="dff_test_vhdl", module="dff_cocotb", toplevel_lang="vhdl")
 
 
+@pytest.mark.skipif(os.getenv("SIM") not in ["questa", "modelsim"], reason="TCL is not supported")
+def test_dff_tcl(tmp_path: pathlib.Path):
+    tcl_file_path = pathlib.Path.joinpath(tmp_path, "dff.tcl")
+    with tcl_file_path.open('w') as tcl_file:
+        tcl_file.write(f'vcom {os.path.join(tests_dir, "dff.vhdl")}')
+    run(tcl_sources=[tcl_file_path], toplevel="work.dff_test_vhdl", module="dff_cocotb", toplevel_lang="vhdl")
+
 if __name__ == "__main__":
     test_dff_verilog()
     # test_dff_vhdl()
```

themperek, 2022-09-06:
I would do this instead by extension

Something like (not tested):

```python3
class QCustom(Questa):
    def __init__(self, tcl_sources, *argv, **kwargs):
        self.tcl_sources= tcl_sources
        super().__init__(*argv, **kwargs)

    def  do_script(self):
         return = f"source {self.tcl_sources};" + super().do_script()

def test_abc():
    QCustom(
        ...
        tcl_sources="abc.tcl"
    )
    
 ```

Would this work for you ?

alexgomp, 2022-10-05:
Yep, that should do the trick. Thank you!

closed 2022-10-05 by alexgomp