themperek/cocotb-test #207: ModuleNotFoundError: No module named 'cocotb._vendor.find_libpython'

aignacio, 2022-11-17:
Hey @themperek,

I'm having this error recently, any clue what's that?

```python
ImportError while importing test module '/axi_dma/tb/test_dma_abort.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tb/test_dma_abort.py:20: in <module>
    from cocotb_test.simulator import run
.tox/py38/lib/python3.8/site-packages/cocotb_test/simulator.py:13: in <module>
    import cocotb._vendor.find_libpython as find_libpython
E   ModuleNotFoundError: No module named 'cocotb._vendor.find_libpython'
```

aignacio, 2022-11-20:
hey @themperek,

can you release the updated branch?

themperek, 2022-11-20:
@aignacio 

>can you release the updated branch?
Done. Hope it works.