themperek/cocotb-test #208: Fix ModuleNotFoundError: No module named 'cocotb._vendor.find_libpython' #207 for  Python <3.7

likewise, 2022-11-21:
Issue #207 was fixed in 0.2.3, but 0.2.3 depends on Python 3.7+.

For Python <3.7 (in my case 3.6.9) I fixed my local installation as follows, in order to work with CocoTB 1.7.2 again:

```bash
sed -i \
-e 's@import cocotb._vendor.find_libpython as find_libpython@import find_libpython@' \
-e 's@install_requires=["cocotb>=1.5", "pytest"],@install_requires=["cocotb>=1.5", "pytest", "find_libpython"],@' \
`find /usr/local/lib/python*/*-packages/cocotb_test/simulator.py`
```
to fix my Ubuntu system-wide installation. The above requires sudo rights.

```bash
sed -i \
-e 's@import cocotb._vendor.find_libpython as find_libpython@import find_libpython@' \
-e 's@install_requires=["cocotb>=1.5", "pytest"],@install_requires=["cocotb>=1.5", "pytest", "find_libpython"],@' \
`find ~/.local/lib/python*/*-packages/cocotb_test/simulator.py`
```
to fix any pip3 user local installation.

themperek, 2022-11-23:
https://endoflife.date/python

themperek, 2023-01-23:
Sorry 3.6 will not work. You can try to support if needed.

closed 2023-01-23 by themperek