themperek/cocotb-test #212: parameter with large number does not get handled correctly

catkira, 2023-01-01:
I have a parameter with a rather long number (because iverilog does not support multidimensional parameters, so I sequeeze multiple numbers into one).
The parameter reaches the hdl code correctly, but something goes wrong on the python side.

The parameter is this number:
`
1043161809724755117294630463171082084379269244686810311029607490900248788420381392348463362674117467351380280716396061994807856922595934162415011370737945306097471762891582355257266129874607407022695562401368170834331191570587525271668692425035152628033539633035671272822809848481969293316809956131050719687308142627180187719859186358020489682827020566615291426464441948895237962125901086180674120729539754342713718442926328439298090536281793869892530559822533699520885460200043828645675500690787021624293704335548279396031745742539130949779391845360041558468656930060946865658437656234172368262891111818743752245284837962427971856480068009631783627803463593158425393489613888769529055843669738695165161575481544351020262523014509861557762884136920067057263401584677476858151676420644240180037273464619824125783108857328600765813075760694118749230749032679295972977002489474784749875908710432315440784621321055372136598051978044337017931726632721237632407266511479457338798037948587694505331102427565179203351786911664035714952736332056963449894161364778566454808486588047682353303007806935225584948894648374367526802728090180560360178179648464611823616681910331503501853221338831442171912737474561170439869566397269273475997939859449
`

However in my Python model the parameter becomes `65529`

You can see the details in the test run of my workflow at https://github.com/catkira/5G_SSB_sync/actions/runs/3816910503/jobs/6492835861
Just unfold "Verify with cocotb and icarus"

The parameter gets assembled here: https://github.com/catkira/5G_SSB_sync/blob/master/tests/test_PSS_correlator.py#:~:text=upper%2016%20Bit-,PSS,)),-extra_env
And printed inside the model here:
https://github.com/catkira/5G_SSB_sync/blob/master/model/PSS_correlator.py#:~:text=)%3A-,print,),-pass

You can see all the versions that I am using the the CI run. The cocotb-test 0.2.4 is not yet on pypi.

@themperek: maybe You can quickly see where the problem is before I dig into cocotb-test

themperek, 2023-01-01:
I guess this is not an `int` here ?https://github.com/catkira/5G_SSB_sync/blob/45874427129b440ed794299b9668608070e83452/tests/test_PSS_correlator.py#L31

catkira, 2023-01-01:
I have added a print() without conversion to int, but it prints the same `65529` number:
https://github.com/catkira/5G_SSB_sync/commit/1b040a5a20991487542861d3a338a4cfda6d7fe0

I think `int()` is not the problem, because `int` in python can have 'indefinite' many numbers.

themperek, 2023-01-01:
Try `dut.PSS_LOCAL.value.binstr` ?

It does not seem `cocotb-test` issues. It is `iverilog/vpi` or `cocotb` itself. I will close here.

You also set it by environmental variable https://github.com/catkira/5G_SSB_sync/blob/45874427129b440ed794299b9668608070e83452/tests/test_PSS_correlator.py#L126 so can get it this way.

catkira, 2023-01-01:
yes, after looking at it a bit more, I also think its cocotb related. binstr does not work.

closed 2023-01-01 by catkira